smoltcp-rs/smoltcp #267: Tokio integration

DemiMarie, 2018-12-19:
It would be very nice to be able to use smoltcp with the Tokio ecosystem.  For me, the first use case that comes to mind is extremely high-performance servers that use kernel bypass networking, but there are probably many, many others.

whitequark, 2018-12-19:
Sure. I am not personally invested in Tokio so I will not work on it myself.

Gymmasssorla, 2019-05-25:
This is probably a good idea, but it's strange that high-level abstractions like futures will be in such a low-level crate.

whitequark, 2019-05-25:
I think futures are pretty low level, it's just another way to represent control flow. In general, microcontroller firmware often greatly benefits from using async/await-like control flow rather than threads; there are many coroutine replacements (not really implementations) for C specifically for this task.

Gymmasssorla, 2019-10-09:
Agree with you.

stlankes, 2020-02-04:
Do you still working on a Tokio integration? Do you have already a test example?

dingelish, 2020-07-18:
very interested in tokio integration as well. do you need help?

whitequark, 2020-07-19:
Sure, would be helpful to have even a proof of concept.

Dirbaio, 2020-12-26:
I'd love to see async/await support in smoltcp too!

From some experiments I've done (#382), I believe the best approach is to:
- Add support in smoltcp to associate `Waker`s to sockets.
- Leave everything else to external higher-level crates.

The higher-level crates would implement wrappers for the smoltcp types that implement the desird async traits (AsyncRead, AsyncWrite, Stream for UDP packets..?).

I believe this is the best way to go, because:

- It is not clear which traits should smoltcp implement:
  - The `std` ecosystem has multiple competing incompatible traits, mainly `tokio` and `futures`.
  - The `no_std` ecosystem hasn't converged on any standard io traits yet. The `tokio` and `futures` traits aren't `no_std` compatible. 
- There are non-trivial tradeoffs when implementing async support: how are sockets allocated, is there a single global tcp/ip stack or multiple, how are lifetimes handled.

Closing this as we won't be integrating with Tokio specifically, but the PR adding `Waker` support is coming very soon! :)

closed 2020-12-26 by Dirbaio