smoltcp-rs/smoltcp #269: httpclient_example: segment not in receive window

kevinis, 2019-01-03:
Environment: VMware Fusion, Ubuntu 18.04, kernel 4.15.0, rust stable 1.29.1

The tap0 interface was set up exactly as the Hosted usage examples section in README.md.
Running the httpclient example kept failing with multiple "segment not in receive window" errors.
```shell
cargo run --example httpclient -- tap0 129.219.108.56 http://sefcom.asu.edu
```

 Below please find the screenshot from Wireshark.

[image]

I also tried the httpclient example on http://example.org. In most cases it received a complete response but it did failed a few times with the same "segment not in receive window" errors.

I'm not sure if this is a duplicate of #97.  At least Wireshark tells a different story where multiple TCP packets are labeled as "TCP Window Full".

whitequark, 2019-01-03:
That's interesting. Could you try again with 556672f79811a1015c98afb329b8e47242fe3f29?

kevinis, 2019-01-03:
Just tried 556672f79811a1015c98afb329b8e47242fe3f29 and v0.5.0. Still the same errors.

whitequark, 2019-01-03:
Interesting. "ACKed unseen segment" looks like a bug, definitely.

kevinis, 2019-01-28:
Hello! Any suggestion on how I can help in this issue?

whitequark, 2019-01-29:
Personally, I would enable the highest log level and then very carefully examine the Wireshark packet traces against the verbose logs. Unfortunately there is no better way applicable in all cases that I'm aware of; TCP stacks are tricky to test automatically.

pothos, 2019-01-30:
I think most times I looked at the smoltcp log message about the segment not being in the receive window, it was because it was a retransmitted packet which was not needed anymore and thus not in the window. Looking at my wireshark and the here posted wireshark labels I have doubts about the quality of the label names. E.g., the packet before the `ACKed unseen segment` label contains data which is allowed to be consumed even if it would just be a segment for the purpose of window probing. Therefore, the ACK is valid from my point of view and the wireshark label misleading.

Dirbaio, 2021-03-24:
I can no longer reproduce this. There's been many TCP improvements since. Testing with google.com and example.com works 100% of the time. URL in OP also works 100% of the time but it's now a redirect to HTTPS.

Closing. Feel free to open new issues if you see bad TCP behavior.

closed 2021-03-24 by Dirbaio