smoltcp-rs/smoltcp #270: Use cfg_if to simplify some cfgs

dlrobertson, 2019-01-15:
The [cfg_if] crate is `no_std` and use of the `cfg_if!` macro could simplify some of the code where we have a pattern like the following:

```rust
#[cfg(something)]
// Do something
#[cfg(not(something))]
// Do something else
```

[cfg_if]: https://github.com/alexcrichton/cfg-if

dlrobertson, 2019-02-01:
Looked at this briefly last night and I ran into https://github.com/alexcrichton/cfg-if/issues/8. This greatly limits the usefulness of `cfg_if` for us, but there still may be locations in the codebase that would benefit from the use of `cfg_if`.

whitequark, 2019-02-01:
Oh yeah pretty sure that was the issue. Maybe cfg-if can be fixed?

dlrobertson, 2019-02-01:
Yeah it would require some reworking, but I don't see why you couldn't also allow a `block` or `expr` token. This is of particular interest to me because the current state of `cfg`s in `src/libcore/ffi.rs` for `VaList` are far from readable, so I'll likely look into this a bit over the next week or so.

Dirbaio, 2022-06-06:
I'm going to close this: it's rare that cfg's are paired in `cfg(something), cfg(not(something))`, other cases aren't made that better by `cfg-if`, and IMO it's nice if all `cfg`s are handled the same.

closed 2022-06-06 by Dirbaio