smoltcp-rs/smoltcp #271: Move to rust 2018 edition

dlrobertson, 2019-01-24:
Get all the latest and greatest by moving to rust 2018 edition.

whitequark, 2019-01-24:
This would be a problem for M-Labs because the ARTIQ Rust is not yet at the version that has the edition.

crawford, 2019-01-24:
Crates can use independent editions. smoltcp can switch to 2018 while the rest of your code stays on 2015. It works quite well in practice (I'm already using a mix of 2018 and 2015 with smoltcp).

jbg, 2019-08-29:
> The problem is that M-Labs has a fork of rustc which is not recent.

(out of interest, how come? Can the changes be contributed upstream?)

Would a PR be accepted for 2018 edition yet, or are you still using an older rustc internally?

whitequark, 2019-08-29:
> (out of interest, how come? Can the changes be contributed upstream?)

No. M-Labs is using OpenRISC, which has a fork of LLVM, fork of Clang, and fork of rustc to support them both. There are currently neither resources nor people to upstream the OpenRISC backend in LLVM so it is unlikely that this will change unless M-Labs migrates to RISC-V.

However, that fork can be upgraded, which might be the simplest solution.

hartytp, 2019-08-29:
> However, that fork can be upgraded, which might be the simplest solution.

Yes, it would be a shame if forcing smoltcp to use an old version of rust put off wider community usage/contributions.

Dirbaio, 2020-12-25:
ping @whitequark Has this changed since? Should we upgrade to Rust 2018 now?

whitequark, 2020-12-25:
Yep, absolutely do go ahead and upgrade.

commit 8d185cdd (2020-12-26): Update to Rust 2018. Fixes #271
commit b5271611 (2020-12-26): Update to Rust 2018. Fixes #271
commit 2a858bc1 (2020-12-26): Update to Rust 2018. Fixes #271
commit 1e40b934 (2020-12-27): Update to Rust 2018. Fixes #271
closed 2020-12-27 by Dirbaio