smoltcp-rs/smoltcp #280: UDP checksum may be optional in IPv4

HeroicKatora, 2019-03-14:
# Issue

When the checksum in a UDP packet is `0`, it should be treated as if the checksum check was successful. Currently, verifier and pretty printer checks this more strictly. Note that this is not true for IPv6, where it is specified that the upper layer checksum must not be elided for UDP.

## References

UDP: https://tools.ietf.org/html/rfc768 (directly above User Interface section)
IPv6: https://tools.ietf.org/html/rfc2460#section-8.1

dlrobertson, 2019-03-14:
I'm pretty sure we already do this in the UDP `Repr` [here](https://github.com/m-labs/smoltcp/blob/master/src/wire/udp.rs#L215-L227). Do you have example code that causes this behavior?

HeroicKatora, 2019-03-14:
That match looks broken to me. Both arms resolve to `Err(Error::Checksum)`, independent of the match or the conditional.

HeroicKatora, 2019-03-14:
Not directly, just noticed it in the packet dump of a custom `Phy` implementation when wrapped in `phy::EthernetTracer`. Thus also hard to reproduce without the physical phy. I'll try to get something reproducible if possible.

HeroicKatora, 2019-03-14:
I realize this bugreport is in an awkward state without reproduction, treat it as a mental note until I get back to it with a reproduction, the linked implementation looks good.

dlrobertson, 2019-03-14:
> That match looks broken to me. Both arms resolve to Err(Error::Checksum), independent of the match or the conditional

True. It probably should not return an error in the first arm

HeroicKatora, 2019-03-14:
> > That match looks broken to me. Both arms resolve to Err(Error::Checksum), independent of the match or the conditional

> True. It probably should not return an error in the first 

It's just more confusing than necessary, as it's a double negation. The control flow could be:

```
match (src_addr, dst_addr) {
    // ... except on UDP-over-IPv4, where it can be omitted.
    #[cfg(feature = "proto-ipv4")]
    (&IpAddress::Ipv4(_), &IpAddress::Ipv4(_)) if packet.checksum() == 0 => (),
    _ => return Err(Error::Checksum)
}
```

HeroicKatora, 2019-03-14:
Minimal reproduction

```
use smoltcp::wire::{IpAddress, Ipv4Address};

fn main() {
    let packet: &[u8] = &[
        0x00, 0x42, 0x00, 0x80, // Src post, dst port
        0x00, 0x08, 0x00, 0x00]; // length, checksum
    let packet = smoltcp::wire::UdpPacket::new_checked(packet).unwrap();
    let verify = packet.verify_checksum(
        &IpAddress::Ipv4(Ipv4Address::new(127, 0, 0, 1)),
        &IpAddress::Ipv4(Ipv4Address::new(127, 0, 0, 2)));
    assert!(verify, "Checksum can be zeroed in IPv4");
}
```

dlrobertson, 2019-03-14:
Ah, so this issue would be to move the conditional checks based on IPv4 into `verify_checksum` instead of in `Repr::parse`.

HeroicKatora, 2019-03-14:
> Ah, so this issue would be to move the conditional checks based on IPv4 into verify_checksum instead of in Repr::parse.

I guess that is ultimately proper, although I didn't really intend it. Reproduction for `parse` as used in `EthernetInterface`, which doesn't require moving the check:

```
use smoltcp::phy::ChecksumCapabilities;
use smoltcp::wire::{IpAddress, Ipv4Address, UdpPacket, UdpRepr};

fn main() {
    let packet: &[u8] = &[
        0x00, 0x42, 0x00, 0x80,
        0x00, 0x08, 0x00, 0x00];
    let packet = UdpRepr::parse(
        &UdpPacket::new_checked(packet).unwrap(),
        &IpAddress::Ipv4(Ipv4Address::new(127, 0, 0, 1)),
        &IpAddress::Ipv4(Ipv4Address::new(127, 0, 0, 2)),
        &ChecksumCapabilities::default());
    assert!(packet.is_ok());
}
```

commit 0b370a15 (2019-04-24): Fix omission of udp checksum

Allow the control flow to continue when the representation is parsed for
ipv4 addresses, instead of returning with an error. Additionally adds a
test to ensure the correct behaviour.

Closes: #280

Closes: #290
Approved by: whitequark
commit 5274c403 (2019-04-24): Fix omission of udp checksum

Allow the control flow to continue when udp parsing encounters an
invalid checksum within an ipv4 packet, instead of returning with an
error. Additionally adds a test to ensure the correct behaviour.

Closes: #280

Closes: #290
Approved by: whitequark
commit 3fec6751 (2019-04-26): Fix omission of udp checksum

Allow the control flow to continue when udp parsing encounters an
invalid checksum within an ipv4 packet, instead of returning with an
error. Additionally adds a test to ensure the correct behaviour.

Closes: #280

Closes: #290
Approved by: whitequark
closed 2019-04-26 by m-labs-homu