smoltcp-rs/smoltcp #282: is it possible to add entire subnets as ip_addrs, to create a server that acts as if there are any number of servers?

neetcod, 2019-03-27:
In the server example, `IpCidr` is used to add `ip_addrs`

https://github.com/m-labs/smoltcp/blob/master/examples/server.rs#L56

But after trying, it only responds as `192.168.69.1`. Can I create a server that responds to any tcp connection destined to any address in  `192.168.69.0/24`? I want it to act as though there are servers listening on `192.168.69.1`, `192.168.69.2`, `192.168.69.3`, etc. I tried `192.168.69.0/24`, but it's just the one address `192.168.69.0`.

[tun2socks](https://github.com/ambrop72/badvpn/wiki/Tun2socks) is a tool that seems to accept tcp connections going to any address, as long as it goes through a tun device. It's similar to what i'd like to do.

whitequark, 2019-03-27:
That's not currently supported. It is also a very niche feature, and I'm hesitant to make everyone who uses smoltcp pay for this in terms of space or complexity.

whitequark, 2019-03-28:
I have an lwip wrapper you could possibly use (although it's not currently published as a separate project), but wouldn't it be easier to fork smoltcp?

neetcod, 2019-03-28:
@whitequark wow, I'd love to try your lwip wrapper, is it published as part of another project? I'm looking for the easiest way to create a wildcard tcp listener on a tun or tap device, judging by code from the iCepa project, the usage is quite simple, but I'm at a loss when there are problems outside the Rust part because I'm totally unfamiliar with C

neetcod, 2019-03-29:
i'm interested, i scrolled through your repositories but couldn't find one that looks like a repository with a lwip wrapper inside

whitequark, 2019-03-29:
It's in this commit: https://github.com/m-labs/artiq/tree/0253e0a89dfaf31a740927ce8f03e83efe4f1318/artiq/firmware

neetcod, 2019-03-30:
thanks a lot! i'm looking into it

whitequark, 2020-04-14:
The issue author deleted their account, closing.

closed 2020-04-14 by whitequark