smoltcp-rs/smoltcp #287: ICMPv4 ignores broadcast echo requests

adamgreig, 2019-04-14:
I'm not sure if this is deliberate or not. The [current code](https://github.com/m-labs/smoltcp/blob/master/src/iface/ethernet.rs#L1245) is:

```rust
        if ipv4_repr.dst_addr.is_unicast() {
            let ipv4_reply_repr = Ipv4Repr {
                // snip...
            };
            Packet::Icmpv4((ipv4_reply_repr, icmp_repr))
        } else {
            // Do not send any ICMP replies to a broadcast destination address.
            Packet::None
        }
```

The comment suggests we're avoiding _transmitting_ broadcast replies, but the test is on `dst_addr`, which is about whether we're _receiving_ broadcast requests. We definitely don't want to send a broadcast reply, but for that we should check `src_addr` is unicast; as it stands I think we would reply to a spoofed broadcast `src_addr`.

We might or might not want to reply to a broadcast request. If we did we'd need to grab a suitable interface IP for the source of the response (we can't use the incoming `dst_addr` as it's broadcast).

In any event I guess this check should probably be changed; I'm happy to make a quick PR for that if you agree.

whitequark, 2019-04-14:
I'm pretty sure that's a typo and the check should be on `src_addr`.

adamgreig, 2019-04-14:
Are you okay with sending replies to broadcast destinations too? I'll add that to my upcoming ipv4 broadcast PR (the test I wrote to check it uses a broadcast ping, hence finding this issue).

whitequark, 2019-04-14:
Responding to broadcast pings is reasonable, I think.

commit 0a21cb60 (2019-04-14): Fix IPv4 broadcast handling

A regression meant broadcast IPv4 packets were ignored. Restore
handling those packets and also reply to broadcast ICMPv4 echo requests.

Closes #287.

Closes: #288
Approved by: whitequark
commit 5334b931 (2019-04-14): Fix IPv4 broadcast handling

A regression meant broadcast IPv4 packets were ignored. Restore
handling those packets and also reply to broadcast ICMPv4 echo requests.

Closes #287.

Closes: #288
Approved by: whitequark
closed 2019-04-14 by m-labs-homu