smoltcp-rs/smoltcp #305: panic decoding DHCP domain name server option

cjbe, 2019-09-10:
Using the current smoltcp master, I get a panic during parsing a DHCP response:
```
panicked at 'index out of bounds: the len is 3 but the index is 3', C:\scratch\smoltcp\src\wire\dhcpv4.rs:749:25
```
This panic is from parsing the DHCP option field OPT_DOMAIN_NAME_SERVER [here](https://github.com/m-labs/smoltcp/blob/1ada3da5aa33917daf891dda2242d7f5beadfb10/src/wire/dhcpv4.rs#L743-L751)

The panic occurs as my DHCP server is broadcasting 4 DNS server addresses, and the code here iterates through the complete list of 4 DNS servers but tries to store the parsed addresses into an array of fixed length 3.

In my case the data in the options field is:
`[163, 1, 74, 6, 163, 1, 74, 7, 163, 1, 74, 3, 163, 1, 74, 4]`
This is valid, as [per the RFC](https://tools.ietf.org/html/rfc2132#page-6) the only requirements are a minimum length of 4 octets, and length a multiple of 4 octets (i.e. there is no maximum length).

whitequark, 2019-09-11:
Sounds like we should add fuzzing for DHCP packets too.

cjbe, 2019-09-11:
@whitequark would you accept a PR to only use the first 3 name servers supplied by the DHCP server? This is not a long term solution, but seems useful in the short term. (This seems to be what the current code is trying to implement at the moment with its fixed array of 3 DNS servers)

whitequark, 2019-09-11:
Yes, with a test for the panic.

commit 15a3e84b (2019-09-30): Only use first 3 DHCP advertised DNS servers.

The OPT_DOMAIN_NAME_SERVER DHCP option field can advertise an unbounded
DNS servers, but we currently use a fixed length array to store these
servers. This commit ensures only the first 3 (highest priority) servers
are used.

Fixes #305.
closed 2019-09-30 by whitequark