smoltcp-rs/smoltcp #315: New release

david-sawatzke, 2019-11-22:
The current release v0.5.0 soon won't compile, when the old borrow checker is removed. master doesn't have this issue. Seems like a good idea to release a new version before this becomes a major hassle.

```
$ git checkout v0.5.0
$ cargo build
...
warning[E0713]: borrow may still be in use when destructor runs
  --> src/socket/ref_.rs:61:9
   |
39 | impl<'a, T: Session + 'a> Ref<'a, T> {
   |      -- lifetime `'a` defined here
...
61 |         ref_.socket
   |         ^^^^^^^^^^^ returning this value requires that `*ref_.socket` is borrowed for `'a`
62 |     }
   |     - here, drop of `ref_` needs exclusive access to `*ref_.socket`, because the type `socket::ref_::Ref<'_, T>` implements the `Drop` trait
   |
   = warning: this error has been downgraded to a warning for backwards compatibility with previous releases
   = warning: this represents potential undefined behavior in your code and this warning will become a hard error in the future
   = note: for more information, try `rustc --explain E0729`
```

david-sawatzke, 2020-01-03:
Rust 1.40.0 is out, the released version doesn't compile anymore

stlankes, 2020-01-04:
Please update the version at crate.io. The current version in the repository seems to work...

whitequark, 2020-01-04:
I'll do this soon.

whitequark, 2020-01-21:
Done.

closed 2020-01-21 by whitequark