smoltcp-rs/smoltcp #323: TcpSocket timeout not respected in SynSent

MabezDev, 2020-01-15:
If a TcpSocket is in SynSent with no means of ingressing or egressing (ethernet unplugged) the socket never leaves this state, even when a socket timeout has been set.

Is this the intented behaivour?

MabezDev, 2020-01-16:
Following up with a little more info.

It appears the only place the the timeout check is ran is is `TcpSocket::dispatch`, on my system when the ethernet is disconnected, the socket in SynSent never gets dispatch called on it, but other sockets (Listening ones) do.

I'll try and create a hosted example later on today

whitequark, 2020-01-16:
I think it's an oversight.

MabezDev, 2020-01-17:
I've managed to put together a hosted example, but it required a few changes to the TapInterface such that when the link is down it won't panic (simulating a real link loss).

Steps to reproduce:

1. `cargo run --example client tap0 xxxx xx` where xxxx xx is an unreachable ip and port
2. Take note of the output, every 5 seconds the socket will transition to closed
3. run `ip link set down tap0` to simulate the link loss
4. Within a minute or too, the socket will get stuck in SynSent indefinitely (or until the link is up again)

Heres the patch you can apply to master to run this.

```diff
diff --git a/examples/client.rs b/examples/client.rs
index a9aaf65..a243476 100644
--- a/examples/client.rs
+++ b/examples/client.rs
@@ -13,7 +13,7 @@ use smoltcp::phy::wait as phy_wait;
 use smoltcp::wire::{EthernetAddress, Ipv4Address, IpAddress, IpCidr};
 use smoltcp::iface::{NeighborCache, EthernetInterfaceBuilder, Routes};
 use smoltcp::socket::{SocketSet, TcpSocket, TcpSocketBuffer};
-use smoltcp::time::Instant;
+use smoltcp::time::{Instant, Duration};
 
 fn main() {
     utils::setup_logging("");
@@ -56,6 +56,7 @@ fn main() {
     {
         let mut socket = sockets.get::<TcpSocket>(tcp_handle);
         socket.connect((address, port), 49500).unwrap();
+        socket.set_timeout(Some(Duration::from_secs(5)));
     }
 
     let mut tcp_active = false;
@@ -70,11 +71,15 @@ fn main() {
 
         {
             let mut socket = sockets.get::<TcpSocket>(tcp_handle);
+            let socket_state = socket.state();
+            info!("Socket state: {:?}", socket_state);
             if socket.is_active() && !tcp_active {
-                debug!("connected");
+                info!("connected");
             } else if !socket.is_active() && tcp_active {
-                debug!("disconnected");
-                break
+                info!("disconnected");
+                // break
+                socket.connect((address, port), 49500).unwrap();
+                socket.set_timeout(Some(Duration::from_secs(5)));
             }
             tcp_active = socket.is_active();
 
@@ -91,12 +96,12 @@ fn main() {
                     (data.len(), data)
                 }).unwrap();
                 if socket.can_send() && data.len() > 0 {
-                    debug!("send data: {:?}",
+                    info!("send data: {:?}",
                            str::from_utf8(data.as_ref()).unwrap_or("(invalid utf8)"));
                     socket.send_slice(&data[..]).unwrap();
                 }
             } else if socket.may_send() {
-                debug!("close");
+                info!("close");
                 socket.close();
             }
         }
diff --git a/src/phy/sys/tap_interface.rs b/src/phy/sys/tap_interface.rs
index f89597f..7623b5d 100644
--- a/src/phy/sys/tap_interface.rs
+++ b/src/phy/sys/tap_interface.rs
@@ -62,7 +62,7 @@ impl TapInterfaceDesc {
         unsafe {
             let len = libc::write(self.lower, buffer.as_ptr() as *const libc::c_void,
                                   buffer.len());
-            if len == -1 { Err(io::Error::last_os_error()).unwrap() }
+            if len == -1 { return Err(io::Error::last_os_error()) }
             Ok(len as usize)
         }
     }
diff --git a/src/phy/tap_interface.rs b/src/phy/tap_interface.rs
index dfdfb27..2a5cb6e 100644
--- a/src/phy/tap_interface.rs
+++ b/src/phy/tap_interface.rs
@@ -98,7 +98,7 @@ impl phy::TxToken for TxToken {
         let mut lower = self.lower.borrow_mut();
         let mut buffer = vec![0; len];
         let result = f(&mut buffer);
-        lower.send(&buffer[..]).unwrap();
+        lower.send(&buffer[..]).ok();
         result
     }
 }

```

MabezDev, 2020-01-17:
> I think it's an oversight.

I tend to agree. If you are able to reproduce this with the above info, what are the steps to take to fix this do you think?

whitequark, 2020-01-21:
Sorry, I don't think I'll have time to do this any time soon. I have a rather large backlog of issues in other projects and not much time.

MabezDev, 2020-01-21:
I understand. I can work around this for now.

MabezDev, 2020-09-18:
Seems this has been fixed by #368 .

closed 2020-09-18 by MabezDev