smoltcp-rs/smoltcp #326: How to integrate into Android VPNService?

lazytiger, 2020-03-04:
Hi, 

I don't know where to ask this question, so I paste it here. When integrated into android, VPNService will provide a fd which represents an interface I assume. How should I do to integrate into smoltcp with it?

whitequark, 2020-03-05:
There are two things in smoltcp that need to change for this to be possible.

First, the [raw socket code here](https://github.com/smoltcp-rs/smoltcp/blob/master/src/phy/raw_socket.rs#L25-L37) should be changed to have a `from_raw_fd` method. This is the easy part.

Second, Android gives you an IP raw socket, and smoltcp currently only works with Ethernet raw sockets. Adding IP interface support has been discussed before (e.g. for SLIP) but it isn't yet done, and unfortunately it's somewhat invasive because the Ethernet and IP layers aren't cleanly separated; likely large parts of `EthernetInterface` will have to be duplicated. This is fairly nontrivial.

(@astro @jordens Looks like the "smoltcp maintainers" team doesn't have access to the repository? I can't seem to be able to label this issue.)

fbzhong, 2020-03-20:
@whitequark Is it possible to encapsulate IP packet to Ethernet frame and send to smoltcp?

whitequark, 2020-03-21:
Yes, you could certainly do that. You would also need to fake ARP responses in that case.

fbzhong, 2020-03-21:
Thank you. Will have a try.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio

gopakumarce, 2021-12-19:
I have succesfully used smoltcp on windows, android, ios and linux - common code running on all these platforms. And everywhere I use Medium::Ip #[cfg(feature = "medium-ip")]

And as for the from_raw_fd, yes thats a seperate layer I have - for my use case I keep it seperate from smoltcp, because I create the sockets myself - I look at the packets myself and filter out what is of interest to me (or not) and then just create a smoltcp packet for that myself, and feed that socket packets using a Device implementation with Medium::Ip in it.