smoltcp-rs/smoltcp #330: recv_slice for RawSocket, IcmpSocket and UdpSocket may lose data

fdb-hiroshima, 2020-03-23:
`recv_slice` for [RawSocket](https://docs.rs/smoltcp/0.6.0/smoltcp/socket/struct.RawSocket.html#method.recv_slice), [IcmpSocket](https://docs.rs/smoltcp/0.6.0/smoltcp/socket/struct.IcmpSocket.html#method.recv_slice) and [UdpSocket](https://docs.rs/smoltcp/0.6.0/smoltcp/socket/struct.UdpSocket.html#method.recv_slice) may lose data if the provided buffer is smaller than the packet to dequeue.

I don't know if it should not dequeue and return an error, as it might not be possible to get a bigger slice when lacking an allocator, or stay with current behavior. However I think the documentation should be clearer as it seems to say nothing about that.

whitequark, 2020-03-23:
Good catch! I seem to have overlooked that while adding these functions. It is possible to use `PacketBuffer::dequeue_with` to return an error in this case (probably `Error::Truncated`).

What we *should* do in this case is an interesting question. The `recv_slice` methods are, essentially, convenience methods, so any limitations they have do not fundamentally alter the capabilities of the socket. If `recv_slice` returns an error and does *not* dequeue, then the caller must handle that case to avoid getting stuck in a receive loop and use `recv` to dequeue the overlong packet. But that seems to defeat the purpose of using `recv_slice`.

Therefore I argue that it should return an error (probably `Error::Truncated`) *and* dequeue the overlong packet, since that's what the caller will have to do in any case where it can use `recv_slice` . In any other case `recv` should be used rather than `recv_slice`.

stlankes, 2023-07-20:
I this still an issue?

whitequark, 2023-07-20:
cc @Dirbaio

whitequark, 2023-11-27:
Fixed by https://github.com/smoltcp-rs/smoltcp/pull/859.

PR #859 (2023-11-27): return RecvError::Truncated when calling recv_slice with a small buffer
When calling `recv_slice`/`peek_slice` in the udp, icmp and raw sockets, data is lost when the provided buffer is smaller than the payload to be copied (not the case for `peek_slice` because it is not dequeued).

With this commit, an `RecvError::Truncated` error is returned. In case of UDP, the endpoint is also returned in the error. In case of ICMP, the IP address is also returned in the error.

I implemented it the way @whitequark proposes it in #330. Data is still lost, but at least the caller knows that the data was tr [... cut]
closed 2023-11-27 by whitequark