smoltcp-rs/smoltcp #331: How can I bind for IP_ADDR_ANY with port 0 to intercept all connections?

yuguorui, 2020-03-27:
Hello, I want to implement a Tun-based VPN with smoltcp, so I need to intercept all connections.

But I didn't find a way to do that......
I have noticed that `TcpSocket::listen` simply returns a `Error::Unaddressable` when found `port==0`.
Or is there any way to inspect the protocol in the traffic to establish the corresponding socket?

whitequark, 2020-03-27:
> I have noticed that `TcpSocket::listen` simply returns a `Error::Unaddressable` when found `port==0`.

Indeed; if you try this on a BSD-derived network stack (on Windows, Linux, macOS...) you'll notice that the OS allocates a random ephemeral port for you in that case, rather than intercepting all connections. Smoltcp cannot allocate ephemeral ports for you because of the way it's structured, so it returns an error.

> I want to implement a Tun-based VPN with smoltcp, so I need to intercept all connections.

I don't understand why you need to use TCP sockets to implement a TUN-based VPN. Could you describe your requirements in more detail?

yuguorui, 2020-03-28:
Thanks for your reply, sincerely.

I want to implement a rust version of tun2socks on windows, which can forward all traffic through a socks/http proxy.

I know there are tools like [Proxifier ](https://www.proxifier.com) for this purpose, but it cannot handle traffic from UWP or WSL2. I want to solve this from the `Internet layer`, so I need a user-level TCP/IP stack like smortcp to handle connections from local programs.

BTW: 
I am considering to add the feature about accepting all incoming connections to this repo.
My basic idea is to detect the status of the TCP connection in the `process_tcp` function in ethernet.rs, and if the status is SYN, add a new connection in the `SocketSet`.
Some structures also need to be changed, such as `InterfaceInner` and `Packet`.
Could you tell me what do you think of this feature?

Thanks.

whitequark, 2020-03-28:
> I am considering to add the feature about accepting all incoming connections to this repo.

I think it would be a very niche feature that does not justify the added maintenance burden, and I would prefer to not have it upstream.

However, smoltcp is quite modular, so you should be able to reuse much of its infrastructure to achieve your goal. You could take a look at `EthernetInterface::poll` to see what you need to do to implement the functionality you want.

yuguorui, 2020-03-28:
Thank for your advice.

closed 2020-03-28 by yuguorui