smoltcp-rs/smoltcp #339: informations about compilation on mobile + capabilities of this stack

lucaszanella, 2020-05-07:
Hi! I want to integrate OpenVPN3 (official C++ OpenVPN client) into my app, which targets Android, iOS, Windows, Linux, macOS. I already did the OpenVPN3 part, I can send and receive IP packets by hand programatically (that is, I don't use sockets at all, I simply call the OpenVPN3 as a library).

Because of that, an userspace TCP/IP stack is needed. The only ones I found for C are ports of DPDK or Linux which are not portable at all. C++ is even worse, I could only find one that was good, but it used a LOT of preprocessing (almost everything in the lib is templated and uses lots of objects composition and unnecessary C++17 things). I found a good one in OCaml, which is the MirageOS TCP/IP stack but compiling OCaml as a library has proven to be very difficult, even more difficult for Android and iOS.

Rust is very C-friendly (I guess) and so I'd be happy to be able to integrate this stack into my project, and I'm even going to release the OpenVPN3 client + the userspace stack as an open source library for anyone to use. 

What I want to know is: 

- Can this stack be easily compiled, as a library with C interface, for the devices I cited? I'm planning to call it from C++ code.

- Can I use this stack in a completely programatically way? I mean, I don't want to open a socket and talk with this stack. I want to be able to feed my IP packets `uint8_t` buffers and the stack will output a buffer with the TCP payload, and vice-versa. In summary: no system calls to open pipes or anything like that

I know I can get this information from the source code, but since they're easy for someone familiar with the stack to respond, and since I don't know Rust yet (but I'm willing to learn once this proves viable in my project), I'm kindly asking here.

Thank you so much and have a great day!

whitequark, 2020-05-07:
> * Can this stack be easily compiled, as a library with C interface, for the devices I cited?

You can compile smoltcp for any Rust target with no native dependencies, but it uses generics fairly extensively so you will encounter some of the same issues as with C++ templates.

> * In summary: no system calls to open pipes or anything like that

I don't understand the rest of your second question, but smoltcp does not perform any system calls on its own, and in fact most of the library is written assuming that there is no OS services, or even an allocator, available at all.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio