smoltcp-rs/smoltcp #343: Understanding packet representations and emit

lucaszanella, 2020-05-24:
I'm working on @Dirbaio's pull request to make an example that uses Tun (in my case a virtual tun, not a file descriptor based tun) since he already separated the IP from Ethernet.

Here are my commits: https://github.com/Dirbaio/smoltcp/compare/ip-interface...lucaszanella:ip-interface?expand=1 I made it compile at least, now I'm working on how to dispatch to the tun interface. 

I've come up with a `dispatch_ip` function that should do what @Dirbaio's`dispatch_ip` does but for a tun device, in my example `tun.rs`, not a tap one (`ethernet.rs` in his). Here's the function: https://github.com/lucaszanella/smoltcp/blob/c2050c3023f5e78e5f6b3dca4d20bba52d90e109/src/iface/tun.rs#L689 or simply:

    fn dispatch_ip<Tx: TxToken>(&mut self, _tx_token: Tx, _timestamp: Instant,
                          _packet: ip::Packet) -> Result<()> {

My question is simple: can I assume that the ip packet `_packet` already has the payload and everything inside it? Then in order to create a buffer with the entire IP packet, I'd do:

```
ip_repr.emit(my_buffer, &caps.checksum);
let payload = &mut frame.payload_mut()[ip_repr.buffer_len()..];
packet.emit_payload(ip_repr, payload, &caps);
```

and then simply send `my_buffer` to the tun interface?

I'm asking because on `dispatch_ip` for `ethernet.rs`:

```
fn dispatch_ip<Tx: TxToken>(&mut self, tx_token: Tx, timestamp: Instant,
                          packet: ip::Packet) -> Result<()> {
        let ip_repr = packet.ip_repr().lower(&self.ip_config.ip_addrs)?;
        let caps = self.state.device_capabilities.clone();

        let (dst_hardware_addr, tx_token) =
            self.lookup_hardware_addr(tx_token, timestamp,
                                      &ip_repr.src_addr(), &ip_repr.dst_addr())?;

        self.dispatch_ethernet(tx_token, timestamp, ip_repr.total_len(), |mut frame| {
            frame.set_dst_addr(dst_hardware_addr);
            match ip_repr {
                #[cfg(feature = "proto-ipv4")]
                IpRepr::Ipv4(_) => frame.set_ethertype(EthernetProtocol::Ipv4),
                #[cfg(feature = "proto-ipv6")]
                IpRepr::Ipv6(_) => frame.set_ethertype(EthernetProtocol::Ipv6),
                _ => return
            }

            ip_repr.emit(frame.payload_mut(), &caps.checksum);

            let payload = &mut frame.payload_mut()[ip_repr.buffer_len()..];
            packet.emit_payload(ip_repr, payload, &caps);
        })
    }
```

it does a series of `emit`s. I don't understand what are these representations and emit things. It looks like it's a way to fill the ethernet packet's payload with the IP stuff then the UDP stuff, for example? 

I'm trying to understand the design of the stack. For example, if the IP packet is an UDP packet, then this packet can be emitted into a buffer. This emit process will first emit the IP header, then will call and emit the UDP payload. Like this:

```
        Packet::Udp((_, udp_repr)) =>
                udp_repr.emit(&mut UdpPacket::new_unchecked(payload),
                              &_ip_repr.src_addr(), &_ip_repr.dst_addr(), &caps.checksum),
```

Why this `emit` design was chosen?

whitequark, 2020-05-24:
This design is necessary to ensure that it is never necessary to buffer a packet before it gets into the PHY buffer. The representation is a compact way to represent the properties of the packet in a way that allows computing its length. The stack asks the PHY to allocate memory for the packet using just the representation, and then writes the packet contents, again using the representation, into the buffe.r

A secondary consequence is that illegal states of the wire-level packet fields cannot be a part of the representation. So the high-level TCP code doesn't have to care about quite as many minor details.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio