smoltcp-rs/smoltcp #348: ManagedSlice bug

lucaszanella, 2020-06-08:
I was using this library and I started having some lifetime problems and I couldnt find out the reason. I posted on reddit (https://www.reddit.com/r/rust/comments/gy7fxy/why_i_cannot_return_this_struct_problem_with/) and somebody clarified to me what was happening.

See:

```
#[derive(Debug)]
pub struct Item<'a, 'b: 'a> {
    socket: Socket<'a, 'b>,
    refs:   usize
}
#[derive(Debug)]
pub struct SocketSet<'a, 'b: 'a, 'c: 'a + 'b> {
    sockets: ManagedSlice<'a, Option<Item<'b, 'c>>>
}

impl<'a, 'b: 'a, 'c: 'a + 'b> SocketSet<'a, 'b, 'c> {
    pub fn new<SocketsT>(sockets: SocketsT) -> SocketSet<'a, 'b, 'c>
            where SocketsT: Into<ManagedSlice<'a, Option<Item<'b, 'c>>>> {
        let sockets = sockets.into();
        Set {
            sockets: sockets
        }
    }

pub struct TunSmolStack<'a, 'b, 'c> {
    sockets: SocketSet<'a, 'b, 'c>,
}

impl<'a, 'b, 'c> TunSmolStack<'a, 'b, 'c> {
    pub fn new(interface_name: String) -> Result<TunSmolStack<'a, 'b, 'c>, u32> {
        let socket_set = SocketSet::new(vec![]);
        Ok(TunSmolStack{
            sockets: socket_set,
        })
    }
```

Here's the full error:

```
error[E0495]: cannot infer an appropriate lifetime for lifetime parameter `'b` due to conflicting requirements
  --> src/virtual_tun/smol_stack.rs:26:12
   |
26 |         Ok(TunSmolStack{
   |            ^^^^^^^^^^^^
   |
note: first, the lifetime cannot outlive the lifetime `'b` as defined on the impl at 16:10...
  --> src/virtual_tun/smol_stack.rs:16:10
   |
16 | impl<'a, 'b, 'c> TunSmolStack<'a, 'b, 'c> {
   |          ^^
note: ...so that the expression is assignable
  --> src/virtual_tun/smol_stack.rs:26:12
   |
26 |           Ok(TunSmolStack{
   |  ____________^
28 | |             sockets: socket_set,
29 | |         })
   | |_________^
   = note: expected  `virtual_tun::smol_stack::TunSmolStack<'_, 'b, 'c>`
              found  `virtual_tun::smol_stack::TunSmolStack<'_, '_, '_>`
   = note: but, the lifetime must be valid for the static lifetime...
note: ...so that the type `std::option::Option<virtual_tun::interface::smoltcp::socket::SocketSetItem<'_, '_>>` will meet its required lifetime bounds
  --> src/virtual_tun/smol_stack.rs:22:26
   |
22 |         let socket_set = SocketSet::new(vec![]);
   |                          ^^^^^^^^^^^^^^
```

turns out, as some guy said in the comments, that the `Managed` lib has a bug that required the `ManagedSlice` to be `'static`: https://docs.rs/managed/0.7.1/src/managed/slice.rs.html#74-78

To be fair I'm still understanding why, because this link points to the `From` trait, not the `Into`, but I did update to the latest `managed` crate from git and the error went away.

So shouldn't you update the managed also?

whitequark, 2020-06-08:
I'll release a new version soon.

whitequark, 2020-06-09:
Done: https://crates.io/crates/managed/0.7.2. This should be semver-compatible with all existing use of smoltcp.

closed 2020-06-09 by whitequark