smoltcp-rs/smoltcp #352: Possible lifetime bug for TcpSocket vs UdpSocket

lucaszanella, 2020-06-11:
I'm new to Rust so I'm still understanding lifetimes. I had a discussion about the TcpSocket::into usage here: https://users.rust-lang.org/t/understanding-an-unconventional-lifetime-error-stuck-on-this-for-days/44093/13 but if you dont want to read, here's the summary:

```
impl<'a, 'b: 'a, 'c: 'a + 'b> TunSmolStack<'a, 'b, 'c> {
    pub fn new(interface_name: String) -> Result<TunSmolStack<'a, 'b, 'c>, u32> {
        let device = TunDevice::new("tun").unwrap();
        
        let neighbor_cache = NeighborCache::new(BTreeMap::new());
        let socket_set = SocketSet::new(vec![]);
        let mut interface = InterfaceBuilder::new(device)
            .neighbor_cache(neighbor_cache)
            .finalize();
        Ok(TunSmolStack {
            sockets: socket_set,
        })
    }

    pub fn add_socket(&mut self, socket_type: SocketType) -> usize {
        match socket_type {
            SocketType::TCP => {
                let rx_buffer = TcpSocketBuffer::new(vec![0; 1024]);
                let tx_buffer = TcpSocketBuffer::new(vec![0; 1024]);
                let socket = TcpSocket::new(rx_buffer, tx_buffer);
                self.sockets.add(Socket::Tcp(socket));      
            }
```

this code was giving this error on line `self.sockets.add(socket);`:

```
error[E0495]: cannot infer an appropriate lifetime for lifetime parameter `'b` due to conflicting requirements
  --> src/virtual_tun/smol_stack.rs:51:30
   |
51 |                 self.sockets.add(socket);        
   |                              ^^^
   |
note: first, the lifetime cannot outlive the lifetime `'b` as defined on the impl at 23:10...
  --> src/virtual_tun/smol_stack.rs:23:10
   |
23 | impl<'a, 'b, 'c> TunSmolStack<'a, 'b, 'c> {
   |          ^^
note: ...but the lifetime must also be valid for the lifetime `'c` as defined on the impl at 23:14...
  --> src/virtual_tun/smol_stack.rs:23:14
   |
23 | impl<'a, 'b, 'c> TunSmolStack<'a, 'b, 'c> {
   |              ^^
note: ...so that the types are compatible
  --> src/virtual_tun/smol_stack.rs:51:30
   |
51 |                 self.sockets.add(socket);        
   |                              ^^^
   = note: expected  `&mut virtual_tun::smoltcp::socket::SocketSet<'_, '_, '_>`
              found  `&mut virtual_tun::smoltcp::socket::SocketSet<'a, 'b, 'c>`

```

turns out its because of 

```
impl<'a> Into<Socket<'a, 'a>> for TcpSocket<'a> {
    fn into(self) -> Socket<'a, 'a> {
        Socket::Tcp(self)
    }
}
```

which generates a `Socket` with equal lifetimes for `'b` and `'c` on my struct.

I noticed `UdpSocket` does not have this problem:

```
impl<'a, 'b> Into<Socket<'a, 'b>> for UdpSocket<'a, 'b> {
    fn into(self) -> Socket<'a, 'b> {
        Socket::Udp(self)
    }
}
```

Is there a reason why you did `Socket<'a, 'a>` instead of `Socket<'a, 'b>`?

For now I'm doing this to avoid the problem:

` self.sockets.add(Socket::Tcp(socket));     `

whitequark, 2020-06-11:
Could be just a bug. Does changing to  `Socket<'a, 'b>` fix your issue?

lucaszanella, 2020-06-11:
Did

```
impl<'a, 'b> Into<Socket<'a, 'b>> for TcpSocket<'a> {
    fn into(self) -> Socket<'a, 'b> {
        Socket::Tcp(self)
    }
}
```

and it seems to work. The intellisense accuses of the same error as before BUT it compiles nicely. It didn't before.

whitequark, 2020-06-11:
Please send a PR.

Dirbaio, 2020-12-26:
This was fixed in https://github.com/smoltcp-rs/smoltcp/commit/6bc6cc7af7ab7b710e9238b3c34bd50425ed7a92

closed 2020-12-26 by Dirbaio