smoltcp-rs/smoltcp #354: Panic due to (invalid?) src address during TCPv4 ingress

dnadlinger, 2020-06-11:
I don't have a dump of the offending packet for this yet, but perhaps the issue is obvious enough without it:

```
/home/bb/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/wire/ip.rs:850:18: Unexpected pseudo header addresses: *, 10.255.6.211
backtrace for software version 5.0.dev+1962.g91dcf10c;master:
```

```
/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1534014507178/work/src/libcore/panicking.rs:92
/home/bb/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/wire/ip.rs:850
/home/bb/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/wire/tcp.rs:344
/home/bb/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/wire/tcp.rs:764
/home/bb/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1394
/home/bb/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/iface/ethernet.rs:1001
/var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1534014507178/work/src/libcore/ops/function.rs:223
```

The issue seems to be that `Packet::verify_checksum` is called on a packet with `Unspecified` `src_addr` rather than `IPv4` like `dst_addr`, and hence panics (which is documented behaviour).

dnadlinger, 2020-06-11:
Possible patch (although there might be a more appropriate place to check this, e.g. at a higher level, or in `verify_checksum` itself):
```diff
diff --git a/src/wire/tcp.rs b/src/wire/tcp.rs
index 8242b36..1ff2699 100644
--- a/src/wire/tcp.rs
+++ b/src/wire/tcp.rs
@@ -1,4 +1,4 @@
-use core::{i32, ops, cmp, fmt};
+use core::{i32, ops, cmp, fmt, mem};
 use byteorder::{ByteOrder, NetworkEndian};

 use {Error, Result};
@@ -760,6 +760,10 @@ impl<'a> Repr<'a> {
         // Source and destination ports must be present.
         if packet.src_port() == 0 { return Err(Error::Malformed) }
         if packet.dst_port() == 0 { return Err(Error::Malformed) }
+        // Address families must match.
+        if mem::discriminant(src_addr) != mem::discriminant(dst_addr) {
+            return Err(Error::Malformed)
+        }
         // Valid checksum is expected.
         if checksum_caps.tcp.rx() && !packet.verify_checksum(src_addr, dst_addr) {
             return Err(Error::Checksum)
```

whitequark, 2020-06-11:
Patch seems fine as a workaround but not an appropriate way to fix this. Please do capture the packet.

dnadlinger, 2020-06-11:
Reproducing this might be hard – I've seen this particular panic once so far in years. Where else would the address families be checked? If `Ip::Repr` allows mismatching families, surely `Tcp::Repr` would need to verify this?

whitequark, 2020-06-11:
It's late today but I'll take a closer look tomorrow.

dnadlinger, 2020-06-11:
Sure, no worries. Thanks!

dnadlinger, 2020-06-11:
Hmm, this looks really strange.

```rust
    #[cfg(feature = "socket-tcp")]
    fn process_tcp<'frame>(&self, sockets: &mut SocketSet, timestamp: Instant,
                           ip_repr: IpRepr, ip_payload: &'frame [u8]) ->
                          Result<Packet<'frame>>
    {
        let (src_addr, dst_addr) = (ip_repr.src_addr(), ip_repr.dst_addr());
        let tcp_packet = TcpPacket::new_checked(ip_payload)?;
        let checksum_caps = self.device_capabilities.checksum.clone();
        let tcp_repr = TcpRepr::parse(&tcp_packet, &src_addr, &dst_addr, &checksum_caps)?;
        …
```

Both are taken immediately from ip_repr.

```rust
    /// Return the source address.
    pub fn src_addr(&self) -> Address {
        match self {
            &Repr::Unspecified { src_addr, .. } => src_addr,
            #[cfg(feature = "proto-ipv4")]
            &Repr::Ipv4(repr) => Address::Ipv4(repr.src_addr),
            #[cfg(feature = "proto-ipv6")]
            &Repr::Ipv6(repr) => Address::Ipv6(repr.src_addr),
            &Repr::__Nonexhaustive => unreachable!()
        }
    }

    /// Return the destination address.
    pub fn dst_addr(&self) -> Address {
        match self {
            &Repr::Unspecified { dst_addr, .. } => dst_addr,
            #[cfg(feature = "proto-ipv4")]
            &Repr::Ipv4(repr) => Address::Ipv4(repr.dst_addr),
            #[cfg(feature = "proto-ipv6")]
            &Repr::Ipv6(repr) => Address::Ipv6(repr.dst_addr),
            &Repr::__Nonexhaustive => unreachable!()
        }
    }
```

The logic here matches nicely.

I really hope this isn't some random memory corruption.

dnadlinger, 2020-06-11:
Judging from the backtrace, the call comes from Ethernet `InterfaceInner::process_ipv4`, so the Repr::Ipv4 case should definitely be hit…

whitequark, 2020-06-11:
Yeah that seems really sketchy. Is IPv6 even turned on in your device?

dnadlinger, 2020-06-11:
It's a standard Kasli build (ARTIQ master), so yes.

whitequark, 2020-06-15:
> I really hope this isn't some random memory corruption.

I think it, unfortunately, is. My two arguments is that (a) this doesn't look possible and (b) we have a fuzzer that should have definitely found this, since it crashes in the packet parser, before there is any chance for any connection state to come into play. If you'd like I can try fuzzing for a few hundred CPU-hours more, otherwise I propose to close it.

dnadlinger, 2020-06-16:
Agreed, it really doesn't look like it should be possible. The bug must be elsewhere (OR1k codegen, memory corruption, …).

closed 2020-06-16 by dnadlinger