smoltcp-rs/smoltcp #362: no data flows on TCP when Raspberry Pi 4 ethernet offloading is enabled

sbourdeauducq, 2020-07-03:
When using a Raspberry Pi 4 with a recent kernel (e.g. 5.6.18) connecting to a smoltcp server, the TCP connection is established but then no data flows. The same behavior is observed on different devices (ARTIQ, STM32).

Attached is a packet capture; the RPI sends the data but it seems the smoltcp server does not process it.

[rpi4smoltcp.zip](https://github.com/smoltcp-rs/smoltcp/files/4870380/rpi4smoltcp.zip)

This can be worked around by running ``ethtool -K eth0 rx off tx off`` on the Raspberry Pi.

Dirbaio, 2020-07-05:
I took a look at the capture. It shows the rpi sends 10 bytes to smoltcp, and smoltcp correctly ACKs it. It seems like the TCP part is working correctly. I agree that there must be something fishy if it works without offloading but not without, but from the capture I don't see what it is.

In the capture, the TCP checksums for tx'd packets is incorrect. AFAIU this is normal if you have checksum offload enabled (the CPU sends the packet to the NIC without calculating the checksum, and the NIC calculates it when sending it. Since you're capturing on the CPU you get the packets without calculated checksum).

The checksum for rx'd packets is correct though, so probably smoltcp is filling them correctly, but maybe the NIC is auto-fixing them, and they're wrong on the wire? 

- what are you passing in `ChecksumCapabilities`? `None` means offload, `Both` means don't offload (calculate in software). If your NIC doesn't support offloading you should use `Both`.
- Just to make sure: the attached capture corresponds to the "not working" case, not the "working" case?
- What do you mean by "the smoltcp server does not process it"? It doesn't get returned from `recv`, `recv_slice`, `peek`, etc? Should the server respond with something that's not hitting the wire?
- What's the complete setup? The smoltcp device is a Linux machine using `tap`, or raw socket, or custom hardware? 
- Attach packet captures for the "working" and "not working" cases, so we can hopefully spot a key difference.
- Attach packet captures from the actual wire (port mirroring, or linux machine with 2 ethernet interfaces as bridge). This way we 
- Attach debug logs from smoltcp (`debug` and `trace` levels)

sbourdeauducq, 2020-07-07:
> * Just to make sure: the attached capture corresponds to the "not working" case, not the "working" case?

That's the not working case of course.

> * What do you mean by "the smoltcp server does not process it"? It doesn't get returned from `recv`, `recv_slice`, `peek`, etc? 

Yes.

> * What's the complete setup? The smoltcp device is a Linux machine using `tap`, or raw socket, or custom hardware?

RPI4 connecting to an embedded smoltcp device (seen on STM32 and ARTIQ). Not tried other configurations with RPI4 as client yet.

Dirbaio, 2020-07-07:
Can you provide capture from the network, and smoltcp trace logs? it'll be impossible to figure this out otherwise.

sbourdeauducq, 2020-07-29:
Seems to have to do with the Raspberry Pi kernel version as well. 5.6.18 provokes the bug, 5.7.8 does not.

occheung, 2020-07-29:
Attached are packet captures on the wire.

[wire-capture.zip](https://github.com/smoltcp-rs/smoltcp/files/4993411/wire-capture.zip)

Connection: RPI4 client connects to STM32 smoltcp server
Intended behavior: When receive a message, send back the cumulated number of character sent immediately.
Server timeout is set to be 2 seconds

The message duplicates itself in the packet when recv is not working. (e.g. packet no. 4, sent "test\n")
recv works for those that don't have duplicated message. (e.g. packet no. 30, sent "nothing\n")

occheung, 2020-07-31:
It seems that there are issues with RPI4 pushing TCP packets onto the wire in 5.6.18.

For me, it only happens when the ethernet frame size is too small and padding is needed (i.e. frame length < 60, not counting frame check sequence). Normally, it is handled by adding paddings with 0-octets to make it at least 60 bytes. For some reason, rpi4 duplicates the data. Below are a few of my observations:

1. This rpi4 issue seems to be related to the TCP checksum.
The TCP checksum of the sent packets seems to be computed by ignoring the duplicate data. Wireshark thinks otherwise.
So for packet 4 of "wire-capture.zip", Wireshark thinks the checksum should be the ```0x2f5d```, probably calculated by adding the ASCII value of ```test\n``` to the received checksum (i.e. ```0x556b+0x0074+0x6573+0x740a = 0x2f5d``` in 1's complement)
Maybe the problem is the checksum? What if the checksum doesn't change after the duplication? The easiest method I could think of is to send ```'\0'```s from RPI4. Here are the captures from the wire.
[5-null-byte-test.zip](https://github.com/smoltcp-rs/smoltcp/files/5004944/5-null-byte-test.zip)
```recv``` works in this case. Only 5 bytes are returned, which is correct.

2. This issue pops up with another non-smoltcp server as well.
I tried to set up a very simple Netcat server on another machine and connected RPI4 as a Netcat client.
host: ```nc -l <port>```
client: ```nc <ip address> <port>```
[image]
(Left side is the client, the right side is the server, only client sent messages, with hidden '\n' sent at the end of each line)
Sending null characters were also tested and yielded same result.

I am not quite sure whatever happened to me is the same as the OP's issue.
- OP sent 10 bytes. No padding will be needed. My findings are strictly related to ethernet paddings.
- The PSH got an ACK. As @Dirbaio mentioned, everything seems fine on the TCP part.

TLDR: On my side, there seems to be an issue with RPI4 handling TCP packet when ethernet padding is needed. The issue happens even when smoltcp devices are not involved. It is probably a RPI4 issue.

PS: Everything is fine when using RPI4 with kernel version 5.7.10

sbourdeauducq, 2020-07-31:
>     2\. This issue pops up with another non-smoltcp server as well.

Okay, I think it seems pretty clear this is a RPI/Linux bug not related to smoltcp.

closed 2020-07-31 by sbourdeauducq