smoltcp-rs/smoltcp #364: What is the default behaviour when there are multiple IP addresses

lucaszanella, 2020-07-21:
Hi, I tried reading the code but it wasn't clear to me what is the default behaviour of when there are multiple IP addresses. For example, if I start a TCP connection, which IP of those will be chosen as the source IP address?

whitequark, 2020-07-22:
> For example, if I start a TCP connection, which IP of those will be chosen as the source IP address?

Right now the [documentation](https://docs.rs/smoltcp/0.6.0/smoltcp/socket/struct.TcpSocket.html#method.connect) doesn't specify it, so you should only expect an arbitrary address to be chosen. (IIRC it's the first one, but since it's not documented, knowing which one it is means you can't rely on it anyway.) If you want a particular one to be used, you can specify it yourself.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio