smoltcp-rs/smoltcp #365: Is the poll/send-receive/phy_wait pattern ok?

lucaszanella, 2020-07-22:
Hi, I'm finishing the writings of the SmolTCP Cpp interface (https://github.com/lucaszanella/smoltcp_cpp_interface) and there's something I need to consider.

On all examples of usage of SmolTCP, I've seen this pattern

```
poll

socket send/receive

phy_wait
```

polling simply takes the contents of the socket buffers and writes to tun/tap, and reads from tun/tap and put data into socket buffers

socket send/receive sends to socket buffer or reads from socket buffer

phy_wait waits for data to be available in the tun/tap device be it for read or write

For me it looks like, after poll, if we send a small ammount of data, let's say a TCP connect request , it'll get into the socker buffer. Then we wait with `phy_wait`, but there's no sense in waiting for anything from the file descriptor because we didn't even send anything yet. Then the `phy_wait` timeouts and `poll` is called again, which sends the connect request we had in the socket buffer.

Should we really do things like this? It doesn't look that `phy_wait` is doing real great here, it simply delays the polling up to the timeout.

For me it makes more sense to first write/read from sockets, then poll to 'commit' those changes, then wait for data with phy_wait, but even then I have my doubts.

whitequark, 2020-07-22:
> On all examples of usage of SmolTCP, I've seen this pattern

IIRC, I didn't put a great amount of thought into it--it was just something to get the examples working.

rickpayne, 2020-08-11:
Any thoughts on how to handle an interface such as a virtio vring where packets arrive asynchronously and should be pushed into the lower part of the stack at that point? In my case, I dont have an fd to select on in the phy_wait style.
I'm fairly new to rust, so probably still have a 'c' mindset, where we get the interrupt, marshall the buffers and then pass them to the next later up with some reference to the source interface...

whitequark, 2020-08-11:
> Any thoughts on how to handle an interface such as a virtio vring where packets arrive asynchronously and should be pushed into the lower part of the stack at that point? In my case, I dont have an fd to select on in the phy_wait style.

It depends on the overall structure of your application. Something as simple as busy-waiting on a virtio empty flag would work, but you probably want to integrate that with scheduling of any other work your application does.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio