smoltcp-rs/smoltcp #366: TcpSocket::connect does not properly handle an already connected socket

ryan-summers, 2020-08-10:
It appears that smoltcp does not properly handle an unexpected ACK when in the SYNC-SENT state. This results in the TCP socket being in an unusable state.

These conditions are described [here](https://serverfault.com/questions/733681/server-sends-ack-in-response-to-syn-causing-a-reset-in-tcp-request-from-the-clie), but the basics are that a previous connection (which was not properly terminated with a FIN) results in the server having some established TCP state.

When the client sends the SYN packet, the server acknowledges with an ACK#x, which is invalid to receive in the SYNC-SENT state. `smoltcp` should recognize this state and instead send a RST to the server to properly terminate the previous connection. It should then restart the SYN handshake to establish a connection with the server.

Sample of wireshark capture showing the error
[image]

ryan-summers, 2020-08-10:
@whitequark I have a fix that basically just `aborts()` the connection if an ACK is received after SYN (without the SYN bit set). The result is that the end user has to re-`connect()` the socket, but keeps the details and statefulness to a minimum in the `smoltcp` TCP implementation. Would we rather implement logic to restart the 3-way handshake after the RST is sent?

whitequark, 2020-08-10:
> but keeps the details and statefulness to a minimum in the `smoltcp` TCP implementation.

I like this approach. If it turns out to be insufficient, we can always go for a more complex one later.

PR #367 (2020-08-17): Adjusting TCP connection 3-way handshake state management
This PR fixes #366 by adding an `abort()` if an `ACK` is received without the `SYN` flag set when in `SYN-SENT`.

The implications of this are that the user will have to re-`connect` the TCP socket after this occurrence.
closed 2020-08-17 by whitequark