smoltcp-rs/smoltcp #370: Permit custom Instant::now() emplementations in no_std environments

IamTheCarl, 2020-08-15:
Instant::now() is dependent on the std crate and therefore disabled when setup to run in a no_std environment. That's understandable.

I have however gotten FreeRTOS running in my no_std environment, which permits me to replicate the behaviour of Instant::now() like so:

```rust
pub fn now() -> Instant {
    Instant::from_millis(FreeRtosUtils::get_tick_count_duration().to_ms())
}
```

Would it be possible to enable this library to link to a user defined Instant::now() function if enabled with a feature? Maybe something that could be tagged in the same way #[panic_handler] works?

whitequark, 2020-08-16:
What would be the point? `Instant::now` is never used within smoltcp itself.

IamTheCarl, 2020-08-16:
It's used in the examples.
It could be useful to provide libraries a portable way to get the current time as well.

In my case I'm experimenting with FreeRTOS+smoltcp integration. I got the DHCP example working, but had to implement the now() function myself. I'd like to make it available to the users in a familiar syntax, making the difference from std environment hardly noticeable.

whitequark, 2020-08-16:
I don't think the complexity of the mechanism you propose (i.e. something similar to `#[panic_handler]`) is justified by the benefit of avoiding change a single line in example code.

closed 2020-08-16 by whitequark