smoltcp-rs/smoltcp #371: DHCP ignores device checksum capabilities

Spadi0, 2020-08-17:
`Dhcpv4Client` ignores the device checksum capabiltities, and always uses the default value for `ChecksumCapabilities`. The code responsible can be seen here:
https://github.com/smoltcp-rs/smoltcp/blob/bdfa44270e9c59b3095b555cdf14601f7dc27794/src/dhcp/clientv4.rs#L133

As far as I can tell, this would basically be a one-line fix:
```rust
let checksum_caps = iface.device().capabilities().checksum;
```

This code breaks DHCP for any device that doesn't use the default checksum capabilities.

birkenfeld, 2020-08-17:
"breaks" is a bit dramatic - didn't it just mean that smoltcp always calculates the checksums by itself?

Spadi0, 2020-08-19:
@birkenfeld 
It depends on how the driver handles checksums - if the driver expects the checksum field to be empty, and it is not, then the calculated checksum becomes invalid. I have tested this on a STM32F429NIHx, and IIRC, this is the case on a lot of embedded MCUs with an ethernet peripheral.

As I would imagine that one of the more popular use cases for smoltcp is embedded systems, that is why I used "breaks"; although, as you said, maybe that word is a bit dramatic.

Either way though, I can't see a reason to not change this code. As there is a comment with a question mark, I presume this was boilerplate used when writing the DHCP module that was intended to be changed before it was merged, but it just slipped through the cracks.

birkenfeld, 2020-08-19:
True - I think @whitequark will be thankful if you make your suggestion into a PR.

PR #373 (2020-10-22): Changed Dhcpv4Client to use device checksum capabilities
Fixes #371 by fetching the checksum capabilities from the `Device`.

Previously, the checksum capabilities were always set to `ChecksumCapabilities::default()`, which could mess with the checksum generation.
closed 2020-10-22 by whitequark