smoltcp-rs/smoltcp #376: Support for defmt logging

birkenfeld, 2020-08-29:
https://github.com/knurling-rs/defmt is a new approach to tracing/logging from embedded devices which dramatically reduces bandwidth use by deferring formatting to the host, and using indexed format strings.

I think this is a very good fit for embedded smoltcp, since other than in low-traffic situations it is currently basically impossible to use smoltcp's (extensive) logging.  With defmt, this might become feasible.

Unfortunately, format strings are not compatible since for compactness, `defmt` format strings need to contain more information about the formatted type (e.g. `{:u8}`).

Since `defmt` isn't meant for non-embedded use, smoltcp would have to switch between them using a feature flag.  The easiest way I could see to support both syntaxes is to use a custom proc macro (`net_trace` and `net_debug` are already custom declarative macros anyway), which outputs two versions of the logging call depending on the feature flag selected (and none if logging is switched off).  However, it's also conceivable for `defmt` to implement that, so I opened https://github.com/knurling-rs/defmt/issues/149 there.

cc @thalesfragoso @adamgreig (stm32-eth developers)

whitequark, 2020-08-29:
> since other than in low-traffic situations it is currently basically impossible to use smoltcp's (extensive) logging

You can write the logs to a ring buffer, and then trigger a snapshot of it through some condition. The utility of this is largely limited to lockups, and while that's nice, I agree with you overall.

> However, it's also conceivable for `defmt` to implement that, so I opened [knurling-rs/defmt#149](https://github.com/knurling-rs/defmt/issues/149) there.

I would much prefer that to happen. A separate crate that implements just the proc macro would also be fine.

birkenfeld, 2020-08-31:
Thanks for the feedback! In any case, I'll wait for defmt to stabilize enough for a crates.io release before I try to implement anything...

Dirbaio, 2020-11-02:
I have a PoC of this here https://github.com/akiles/smoltcp/commit/f4cc2c956bfa6db5fea4cdc79078fa0b6bdc9db0

It's quite ugly, and it required changing all `{}` to `{:?}`. It builds for both `defmt` and `log` though!

The ideal solution would be some macro that takes a "supercharged" format string and converts it to a `defmt` format string or a `log` format string depending on enabled features...

Dirbaio, 2021-04-01:
Done in #455! With defmt 0.2 there's no longer a need to change `{}` into `{:?}`

PR #455 (2021-04-01): Add defmt logging support
This adds basic support for optionally logging with `defmt` instead of `log`, controlled via Cargo features.

Any struct that can implement `defmt::Format` now does so. There's a few left, due to:
- Missing impls for Cell, RefCell that should be added in `defmt`.
- Missing impls for `managed` types.

The left structs are not used in log statements anyway, and are usually "big" structs, like sockets. Since this is already fully functional for integration in a firmware that uses `defmt` I would merge this now, and add the missing impls later when they're unblocked.

Fix [... cut]
closed 2021-04-01 by Dirbaio