smoltcp-rs/smoltcp #379: cargo build --target aarch64-apple-ios ,   failed

TongxiJi, 2020-09-26:
error[E0425]: cannot find value `BIOCSETIF` in this scope
  --> /Users/tonnyji/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/sys/bpf.rs:66:29
   |
66 |         try_ioctl!(self.fd, BIOCSETIF, &mut self.ifreq);
   |                             ^^^^^^^^^ not found in this scope
   |
help: consider importing this constant
   |
1  | use libc::BIOCSETIF;
   |

error[E0425]: cannot find value `BIOCIMMEDIATE` in this scope
  --> /Users/tonnyji/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/sys/bpf.rs:94:29
   |
94 |         try_ioctl!(self.fd, BIOCIMMEDIATE, &mut bufsize as *mut libc::c_int);
   |                             ^^^^^^^^^^^^^ not found in this scope
   |
help: consider importing this constant
   |
1  | use libc::BIOCIMMEDIATE;
   |

error[E0425]: cannot find value `BIOCGBLEN` in this scope
  --> /Users/tonnyji/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/sys/bpf.rs:95:29
   |
95 |         try_ioctl!(self.fd, BIOCGBLEN, &mut bufsize as *mut libc::c_int);
   |                             ^^^^^^^^^ not found in this scope
   |
help: consider importing this constant
   |
1  | use libc::BIOCGBLEN;
   |

error[E0425]: cannot find value `BPF_HDRLEN` in this scope
   --> /Users/tonnyji/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/sys/bpf.rs:108:35
    |
108 |             if len == -1 || len < BPF_HDRLEN as isize {
    |                                   ^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `BPF_HDRLEN` in this scope
   --> /Users/tonnyji/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/sys/bpf.rs:116:25
    |
116 |                 &buffer[BPF_HDRLEN] as *const u8 as *const libc::c_void,
    |                         ^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `BPF_HDRLEN` in this scope
   --> /Users/tonnyji/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.6.0/src/phy/sys/bpf.rs:117:23
    |
117 |                 len - BPF_HDRLEN,
    |                       ^^^^^^^^^^ not found in this scope

   Compiling digest v0.8.1
   Compiling crypto-mac v0.7.0
   Compiling block-buffer v0.7.3
   Compiling mio-uds v0.6.7
   Compiling rand_chacha v0.2.1
   Compiling openssl v0.10.30
error: aborting due to 6 previous errors

For more information about this error, try `rustc --explain E0425`.
error: could not compile `smoltcp`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
make: *** [build] Error 101

TongxiJi, 2020-09-27:
add target_os="ios",  搞定

closed 2020-09-27 by TongxiJi

623637646, 2023-01-13:
May I know where did you add the `target_os="ios"`? @TongxiJi , Did you add it into `smoltcp`'s source code?