smoltcp-rs/smoltcp #386: Why not iface.poll(single_socket: &mut Socket) instead of iface.poll(socket_set: &mut SocketSet)?

lucaszanella, 2020-11-02:
The only time the stack interacts with the sockets is in the call

` iface.poll(&mut sockets, timestamp)`.

Before that, I can write and read from the sockets in parallel, then I poll them periodically.

The good thing about this design is that I can parallelize the socket writing and reading. 

However, all the sockets must be contained in a set, the `SocketSet`, so I can poll them with `poll(socket_set: &mut SocketSet)`.

Wouldn't it be better to poll sockets individually by reference?

Suppose I'm writing a client for web cameras. Each one uses one socket. Currently, I have to periodically lock all sockets at once so I can poll them. Even worse: they all have to be in the `SocketSet` (by value, not reference). If they were as reference, at least I could put them momentarily in `SocketSet` just for polling.  

Anyways, if there were a method `iface.poll(socket: Socket)` then I could poll sockets individually as soon as they were free, without having to lock all of them at once (which is hard, will certainly waste time by waiting for some socket to finish being read/written). I could write a loop that polls sockets if they are free, and that waits on a condvar that is unlocked if any of the sockets gets written/read to. 

I think technically I could create multiple `SocketSet` and put 1 socket only inside of each one and poll each of these `SocketSet`s individually, but I did not go that far to test if everything works out. Would this be a good workaround? Shouldn't it be better to have the option to poll individual sockets?

whitequark, 2020-11-02:
Please take a look at the discussion in #385.

Dirbaio, 2020-12-25:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2020-12-25 by Dirbaio