smoltcp-rs/smoltcp #391: Add no_std CI

Dirbaio, 2020-12-24:
#390  added GHA CI except for no_std because of tests not working.

Tests can't run on no_std, but we should at least check the crate builds.

adamgreig, 2020-12-24:
The current GHA workflow does run `cargo check` on a feature set that doesn't have `std` or `alloc`, which is equivalent to checking it builds (we could use `cargo build` instead I guess).

The reason the tests can't run without a std build is lots of the tests include items that are cfg-gated to only exist with either `std` or `alloc`, so the build fails, but in principle if we cfg-gated those tests as well there's no reason you couldn't run the test suite without `std` or `alloc` features. I don't think it would really test anything new, though, so not sure what the value would be.

Dirbaio, 2020-12-24:
My bad, you're absolutely right. I didn't notice the `check` didn't have `std`. :( 

Sorry!

closed 2020-12-24 by Dirbaio