smoltcp-rs/smoltcp #412: New crate.io release?

stlankes, 2021-01-11:
The latest crates.io release (version 0.6.0) is from Januar 2020. Are there any plans to release a new version?

whitequark, 2021-01-12:
At the moment only I have crates.io permissions. @Dirbaio could you set up a team in the GitHub org that I could add to crate owners?

Dirbaio, 2021-01-16:
[done](https://github.com/orgs/smoltcp-rs/teams/crates-io)!

whitequark, 2021-01-19:
Hm, I'm having trouble with that crates.io interface again. It is so horribly unusable.
[image]

stlankes, 2021-01-19:
I thought that `cargo owner --add github:smoltcp-rs:crates-io` is the way to do it.

whitequark, 2021-01-19:
That produces the exact same error.

stlankes, 2021-01-19:
Hm, does crates.io has oauth access to smoltcp-rs?

whitequark, 2021-01-19:
Thanks for the pointer. Done now ([smoltcp](https://crates.io/crates/smoltcp), [managed](https://crates.io/crates/managed)).

whitequark, 2021-01-19:
For the onlookers, what I had to do was to revoke permissions for crates.io for my account, then re-login while granting access to smoltcp-rs. I have no idea whether that's documented anywhere other than in https://github.com/rust-lang/cargo/issues/5297.

whitequark, 2021-01-19:
Actually, there's no release yet, I just set up the permissions, so reopening this.

Dirbaio, 2021-01-19:
smoltcp 0.7 is now out! :tada:

closed 2021-01-19 by Dirbaio