smoltcp-rs/smoltcp #414: Using smoltcp as the network stack in userspace?

ratnadeepb, 2021-01-20:
I am a little confused here. My initial assessment was that I could use smoltcp to implement a network stack that gets packet through the DPDK library. I have created the base version that gets the packets and can transmit the packets. But I couldn't figure out how I go about implementing a device in userspace?

Dirbaio, 2021-01-20:
Yes, you can use smoltcp as a full network stack, that is exactly its purpose.

The overall steps to do this are:
- Implement the `phy::Device` trait with DPDK. When smoltcp wants to send a packet, give it to DPDK. When DPDK has a packet for us, give it to smoltcp.
- Create some sockets (`TcpSocket`, `UdpSocket`, etc) and add them to the SocketSet
- Create a main loop that polls the EthernetInterface and uses the sockets to send/receive data. See the examples.

I'm not sure what you mean by *device*. Do you mean a Linux network device? The smoltcp stack doesn't appear as a normal *Linux network device*. You can't use it with Linux programs that are written for the Berkeley socket API. You have to use the smoltcp as a library in the process where you want to send/receive data, and use its socket API instead of the Berkeley socket API.

ratnadeepb, 2021-01-20:
Yeah. I am confused about the first step really. I see the example of implementing `phy::Device` here https://docs.rs/smoltcp/0.7.0/smoltcp/phy/index.html. I am using two `crossbeam::queue::ArrayQueue`s to store the incoming and outgoing packets. I was wondering how I could implement the `Device` trait for that.

ratnadeepb, 2021-01-20:
I think what is really throwing me off is this part `F: FnOnce(&mut [u8]) -> Result<R>`. I am guessing that it should come from DPDK or anything that receives and sends packets from and to the underlying data structure but could not verify that.

Dirbaio, 2021-01-20:
That's because smoltcp will never allocate buffers for you. The design allows you to choose how they're allocated.

Receiving works like this:
- smoltcp calls Device::receive().
- You have to check if there's a received packet ready AND there's a buffer available for transmitting. Return the tokens if so, return None if not.
- smoltcp calls RxToken::consume()
- You have to call `f` with the packet, that's when smoltcp will process it

Transmitting works like this:
- smoltcp calls Device::transmit()
- You have to check if there's a buffer available for transmitting. Return the tokens if so, return None if not.
- smoltcp calls TxToken::consume() with the packet length `len`
- you have to call `f` with a slice of length exactly `len`. Smoltcp will write the packet to it.
- when `f` returns, you have the packet bytes in the buffer, give it to DPDK.

Dirbaio, 2021-01-20:
How you allocate the buffers is up to you, smoltcp only needs borrowed slices out of these buffers. The simplest way is probably to use a Vec. You can also use statically-allocated fixed-size pools if you want more predictable RAM usage.

ratnadeepb, 2021-01-20:
So I am responsible for 1. getting the packet into and out of the buffers and 2. providing `f`.

Dirbaio, 2021-01-20:
No, smoltcp calls `consume` on the tokens giving you an `f`, you have to call it.

ratnadeepb, 2021-01-20:
Uh! Thanks for the help!

closed 2021-01-20 by ratnadeepb

pothos, 2021-01-20:
I implemented some devices here a while ago, and maybe a look at the netmap code can help even if many things will be different with DPDK:
https://github.com/ANLAB-KAIST/usnet_devices