smoltcp-rs/smoltcp #415: artiq-zynq performance regression with 0.7.0

sbourdeauducq, 2021-01-31:
See https://git.m-labs.hk/M-Labs/artiq-zynq/issues/127

Dirbaio, 2021-01-31:
Interesting. 
Is "H" a Linux machine and "D" a smoltcp device? Assuming so:

- H2D could be due to Delayed ACK if the rx buffer in the device is small. Can you try disabling it? 
- D2H could be due to #387 (but unlikely because the rx buffer in the Linux side is probably huge), or maybe #406 if the link is lossy (the RTT estimation will probably detect a higher RTO, the previous hardcoded RTO of 100ms was very agressive?). 

Could you provide packet captures of before and after? It would help significantly in figuring it out

astro, 2021-02-08:
Regression occurs since the merge of #404.

dnadlinger, 2021-02-08:
@Dirbaio: Just to confirm your interpretation, H would indeed have been a Linux box ("host") running the kernel TCP stack, and D the embedded ARM device running smoltcp.

Dirbaio, 2021-02-08:
> Regression occurs since the merge of #404.

Okay, makes sense. 

- How is the workload like? Is it one big transfer in one direction, or many small transfers in request/response fashion? What's the TCP buffer sizes, what's the size of the writes?
- Can you provide packet capture?
- Have you tried disabling delayed ACK? `socket.set_ack_delay(None)`

dnadlinger, 2021-02-08:
(@astro/… can hopefully provide details, as I'm only peripherally involved with this.)

astro, 2021-02-08:
The workload sizes are 1MB, and 1KB, echoed in a request/response fashion. TCP buffer sizes are 64KB.

`socket.set_ack_delay(None)` mitigated the regression for us. A performance regression in the default settings is still worth investigating, so don't close this issue yet.

astro, 2021-02-25:
I've done some loopback tests with delay and `FaultInjector` in an std environment but was not yet able to reproduce throughput differences between running with and without delayed acks.