smoltcp-rs/smoltcp #417: Can we build `IP over Serial` by the project?

rise0chen, 2021-02-03:
[image]
It's amazing if we can realize this idea.

Dirbaio, 2021-02-03:
This is definitely possible :)

What particular "IP over Serial" protocol are you looking for? Most common are PPP or SLIP. They transmit IP packets instead of Ethernet frames, so you will need #401 . For PPP you can use [this](https://github.com/akiles/ppproto) (beware, it's a bit WIP). SLIP is really simple, I'm sure there are some crates around.

If you don't care about compatibility with a particular protocol and want to use a released version of smoltcp (instead of #401 ) you can also send Ethernet frames over serial (with SLIP, COBS or other framing of your choice).

rise0chen, 2021-02-03:
Both PPP and SLIP need replace `end char` by two other char, so it will need too much cost.
Maybe need another phy layer standard.

rise0chen, 2021-02-03:
Maybe can send Ethernet frames ,and split two frames by some idle time.

pothos, 2021-02-03:
Or parse the Ethernet header to get its length, assuming L2 headers are trusted.

Dirbaio, 2021-03-23:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2021-03-23 by Dirbaio