smoltcp-rs/smoltcp #422: Question: "Fewest-copy" UDP transmission

dnadlinger, 2021-02-18:
A common use case for networking in embedded DSP/DAQ-type applications is to just blast out a data stream via UDP to a host PC, where more intensive and/or offline processing can then take place.

To achieve (close to) line rate in this scenario without cutting too deeply into the real-time CPU budget, it is desirable to avoid extraneous copies on the way to the adapter DMA interface. Since message sizes/order is perfectly predictable in this scenario, ideally one saves the data streams directly into the payload part of a packet ring buffer that is later DMA'd to the adapter by the networking stack.

What is the closest thing to this currently achievable in smoltcp? There is `UdpSocket::send`, but that seems to directly enqueue a buffer, rather than reserving buffer space to be committed later once the message is complete.

astro, 2021-02-19:
It would mean that for every `UdpSocket::send` smoltcp would need to be able to obtain a handle on the next buffer in the ring instead of doing that through the `dispatch()` chain.

It may be easier to reuse smoltcp's `wire` data structures to build packets yourself.

Dirbaio, 2022-06-06:
Closing due to inactivity. If you have more questions, feel free to open new issues!

closed 2022-06-06 by Dirbaio