smoltcp-rs/smoltcp #425: question: listen on 'any' port ?

gopakumarce, 2021-02-25:
So I was using google netstack gvisor tcp/ip in 'promiscuous mode' (thats their terminology) where it will listen and accept connections on 'any' port - which is extremely useful, and is what i need. I see that in  smoltcp we need to specificy individual ports to listen/accept, on a quick scan through the source I did not find any way to listen on any port, can you please tell me if there is any way ? 

Rgds,
Gopa.

gopakumarce, 2021-03-02:
So it took me a while to understand the general usage of smoltcp. IMHO the examples are a lil easier to understand if it were not hiding the details of socket.process() and socket.dispatch() underneath iface.poll(). So if I understand, this is how it works for the case of smoltcp as a server

1. create a socket with TcpSocket::new()
2. do socket.listen(<port>)
3. Feed ip packets to it by calling socket.process() and if socket.process() returns something, transmit that IP pkt out of an interface
4. Periodically call socket.dispatch() to send data thats ready to be sent in the socket
5. When there is data to send out of the socket call socket.send(), periodically call socket.recv() to receive data from the socket

So for my use case of sockets on "all ports", basically in step 3 when I get an IP packet, I will have to maintain some data structure of my own which says "have I seen this five tuple before ? if not, create a new socket and then feed it packets"

gopakumarce, 2021-03-03:
Interestingly if I want to call the process() and dispatch() myself without using the ethernet module, then the module "ip" needs to be public, right now its only crate level visible

whitequark, 2021-03-03:
Those functions are implementation details which are intentionally not exported in the public API.

gopakumarce, 2021-03-03:
that makes sense sure. But i am trying to use smoltcp without ethernet - I want to feed IP packets and transmit IP packets myself, so I am trying to figure out what is the best way to do it in that case. I guess I have to somehow call process() and dispatch() on the socket, but that needs all the IP Repr and Address and stuff right ?

whitequark, 2021-03-03:
There's an in-progress PR #401 that adds support for that.

gopakumarce, 2021-03-03:
Thx! I am checking out the medium-ip-part2 branch now, will try pointing my cargo to that branch and see how that goes

gopakumarce, 2021-03-03:
The one thing I think is something to change  is the socket set passed to the poll - so if I understand right, a "socket" is the complete five-tuple - so if I have like port 80 and hundred flows connecting to port 80 (in server mode), then the poll() routine will basically walk through a list of hundred sockets. Maybe make that a HashMap ? I can volunteer to do that if you think its fine

whitequark, 2021-03-03:
A HashMap wouldn't work due to the way smoltcp is structured internally; it currently has to call a method of every active socket, e.g. in case that socket has had data pushed into its buffers. Changing that would be a major undertaking that completely changes how smoltcp works and it's not even clear if it is desirable.

gopakumarce, 2021-03-03:
ah but then I remember the goal is not to have a heap allocation .. hmm that makes it tricky. I want to use the stack in a case where  will have say two hundred flows/five-tuples/sockets - and for every packet, iterating two hundred sockets wont be too good unfortunately

whitequark, 2021-03-03:
Then smoltcp might not be a good fit for you. It is not trying to be a one-size-fits-all TCP/IP stack but specifically focuses on small embedded targets.

gopakumarce, 2021-03-03:
well smoltcp is a good fit, I think all I need is to expose the socket stuff to the user rather than tying  it to an interface implementation - let the user create and feed packets to the socket, why restrict them to go via an interface / polling structure!

gopakumarce, 2021-03-03:
and by the way, I came here / to smoltcp / rust from golang / gvisor / netstack which was a MONSTER in terms of memory usage. I am trying to write a vpn client on ios which has a 15Mb limitation, and golang / gvisor just wouldnt cut it. And I am EXTREMELY glad that your design requirement is to run on a small embedded system!

Dirbaio, 2021-03-03:
> then the poll() routine will basically walk through a list of hundred sockets. Maybe make that a HashMap ? I can volunteer to do that if you think its fine

Yes, this is a limitation of the current design: poll and dispatch are O(n) I'd really love to see this implemented!

Something like

- Keep a hashmap of sockets, use it for lookup on ingress.
- Keep track of which sockets are "dirty", only poll those on egress.

However

- As @whitequark said, it must work without `std` or `alloc`. This rules out the regular `std::collections::HashMap`, but it's not a blocker, there are [alternatives](https://docs.rs/heapless/0.6.1/heapless/struct.IndexMap.html).
- It probably should be *optional*, via a Cargo feature? Code for hashing and looking up 5-tuples will be quite bigger than an array search, which will hurt code size in embedded applications that only need 1-2 sockets. I don't think the compiler would be able to optimize away the hash/lookup for length-1 hashmaps.

whitequark, 2021-03-03:
> * Keep track of which sockets are "dirty", only poll those on egress.

This is quite tricky though because of timeouts and the fact that anyone can go and poke a socket whenever they want. The current solution is correct in the sense that it guarantees delivery of pending data. I would be hesitant to merge any optimization (and what we're discussing here is purely an optimization) that compromises that guarantee.

whitequark, 2021-03-03:
> * This rules out the regular `std::collections::HashMap`, but it's not a blocker, there are [alternatives](https://docs.rs/heapless/0.6.1/heapless/struct.IndexMap.html).

`managed` has maps: https://github.com/smoltcp-rs/rust-managed/blob/master/src/map.rs

gopakumarce, 2021-03-03:
I actually really really appreciate the design goals here, I got burnt with stupid golang / netstack very bad and anytime I see anyone using more mem than needed I cringe, so I love smoltcp for that. 

And all i think has to be done is to not force the user to go via an "interface" - thats upto the user, provide the tcp stack but let user figure out how to do the interface and polling and all that -let user create a hashmap of sockets (which is what I plan to do) if they have enough memory for that, or let user create a bitmap or whatever if they dont have memory.

in fact, what i am thinking is like this - in the medium-ip-part2, just expose a version of poll() for ONE SOCKET - like poll_one(...) - and I guess thats what people can use in my use case ?

whitequark, 2021-03-03:
> just expose a version of poll() for ONE SOCKET - like poll_one(...)

Hm, can't you already do that though? Use one socket per socket set, then you get `poll_one` automatically.

gopakumarce, 2021-03-03:
ah thats dumb of me. Yes so I can just do the below

1. Create a single interface
2. Maintain my own socket set - ill end up using a hashmap, I am not "that" memory constrained on ios (15mb)
3. When a new packet comes in, parse the five tuple myself and create a new socket and add it to my hashmap
4. And for each packet, find the socket myself, and call iface.polll(<one socket>) 
5. I wont use the iface.poll_delay etc.. (from the examples) since I control the polling and frequency etc.. myself

Sounds like that should work, ill give that a spin, thx!

Dirbaio, 2021-03-03:
> `managed` has maps: https://github.com/smoltcp-rs/rust-managed/blob/master/src/map.rs

True. They need alloc though, maybe we're OK with that? I'd love to have fast dispatch without alloc, but we can always add that later.

> well smoltcp is a good fit, I think all I need is to expose the socket stuff to the user rather than tying it to an interface implementation - let the user create and feed packets to the socket, why restrict them to go via an interface / polling structure!

In the end, no matter what you do, the information from all sockets has to be "joined" together **somewhere** to emit packets. In smoltcp, this "somewhere" is the EthernetInterface. 

You don't "see" that "somewhere" in Go netstack, because it heap-allocates everything, and GC allows it to keep multiple pointers to the same socket in multiple places. One from their "Interface" equivalent, one from user code, done!

However, the equivalent in Rust would be using `Rc` or `Arc`, but that would require `alloc`. We can't do that :)

Instead, you can build your own wrapper that works like that *on top of* smoltcp. For example:

1. Put the SocketSet in a global `static` behind a mutex https://github.com/akiles/embassy-net/blob/9bee576fd241f019c363919b0c29551c6b8ee4b2/embassy-net/src/stack.rs#L38-L47
2. Create wrapper types for each socket type that only have the SocketHandle. For every operation, lock the mutex, reach into the SocketSet to get the real socket and do it. https://github.com/akiles/embassy-net/blob/9bee576fd241f019c363919b0c29551c6b8ee4b2/embassy-net/src/tcp_socket.rs

Now the Socket wrapper "looks" completely independent from the Interface/SocketSet and can be sent around by user code, even to other threads :)

Note that there *even the Mutex* is platform-specific (std's Mutex is not available on embedded targets). This is the reason why smoltcp itself doesn't do this. It has to stay unopinionated so it can work everywhere.

gopakumarce, 2021-03-03:
awesome .. that makes sense. So I will still need  medium-ip-part2 branch for the ip-only interface, hopefully that gets into master sometime soon, till then ill just point cargo to that

gopakumarce, 2021-03-03:
>> You don't "see" that "somewhere" in Go netstack, because it heap-allocates everything

By the way go netstack also has the concept of a "link" and they have ethernet / fd-based links etc.., similar to the interface/iface here

Dirbaio, 2021-03-23:
Hello! As part of issue cleanup I'm closing inactive or already-answered "question" issues, since they're not actionable.

Note this doesn't mean questions aren't welcome. They still are! It's just that they shouldn't stay open forever.

closed 2021-03-23 by Dirbaio