smoltcp-rs/smoltcp #428: 0.7 build error

jiayihu, 2021-03-03:
I've just updated to `0.7` but I cannot build anymore:

```
error[E0004]: non-exhaustive patterns: type `(&Cidr, &Cidr)` is non-empty
   --> /Users/jiayihu/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.7.0/src/wire/ip.rs:334:15
    |
334 |         match (self, subnet) {
    |               ^^^^^^^^^^^^^^
    |
    = help: ensure that all possible cases are being handled, possibly by adding wildcards or more match arms
    = note: the matched value is of type `(&Cidr, &Cidr)`
```

These are the enabled features:

```
smoltcp = { version = "0.6.0", default-features = false, features = ["socket-udp", "socket-icmp"]}
```

wbuck, 2021-03-16:
Yeah, when trying to use version `0.7` of the library I'm getting the following build errors:

    error[E0392]: parameter `'b` is never used 
    --> /Users/wbuck/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.7.0/src/socket/set.rs:182:21
        |
    182 | pub struct Iter<'a, 'b: 'a> {
        |                     ^^ unused parameter
        |
        = help: consider removing `'b`, referring to it in a field, or using a marker such as `PhantomData`

    error[E0392]: parameter `'a` is never used
      --> /Users/wbuck/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.7.0/src/socket/set.rs:13:17
       |
    13 | pub struct Item<'a> {
       |                 ^^ unused parameter
       |
       = help: consider removing `'a`, referring to it in a field, or using a marker such as `PhantomData`

    error[E0392]: parameter `'a` is never used
      --> /Users/wbuckley/.cargo/registry/src/github.com-1ecc6299db9ec823/smoltcp-0.7.0/src/socket/mod.rs:84:17
       |
    84 | pub enum Socket<'a> {
       |                 ^^ unused parameter
       |
       = help: consider removing `'a`, referring to it in a field, or using a marker such as `PhantomData`

Current features:
 `smoltcp = { version = "0.7", default-features = false, features = ["ethernet", "proto-ipv4"] }`

whitequark, 2021-03-17:
@Dirbaio what should we do here, branch and release 0.7.1 with that fixed, or go straight to 0.8?

Dirbaio, 2021-03-23:
I'll start a `v0.7.x` branch with all the non-breaking fixes. The NDISC one is also release-worthy.

closed 2021-03-24 by Dirbaio

Dirbaio, 2021-03-27:
v0.7.1 is now out, with the fixes for feature combination issues!

PR #447 (2021-03-24): Fix feature-related compilation issues.
The goal is that every combination of features either:
- compiles cleanly, or
- fails to compile with a human-readable error message

I've opted to make "no ipv4 and no ipv6" an unsupported combination. It *could* be made to work but I don't think it's worth it. 

Fixed cases: 

- `(no features)` -> now fails with `You must enable at least one of the following features: proto-ipv4, proto-ipv6`. Fixes https://github.com/smoltcp-rs/smoltcp/issues/428#issue-821009561
- `proto-ipv4,socket` -> now fails with `If you enable the socket feature, you must enable  [... cut]