smoltcp-rs/smoltcp #432: question: why does tcp send_slice return bytes == 0 even if send capacity > slice length

gopakumarce, 2021-03-08:
So I call socket.send_slice(&my_slice[..]) - my_slice[...] is like 500 bytes and if I call socket.send_capacity() it returns like 2048 .. But the send_slice still returns Ok(0) .. Suggestions on why that is ? So I was planning to rely on send_capacity() to kind of have a 'foresight' on whether my send would succeed or not, so I guess i should not rely on that ?

Dirbaio, 2021-03-08:
`send_capacity()` returns the *total* capacity of the send buffer, not the *free* capacity. It's essentially the capacity you created the socket with, it doesn't change during operation.

You can use `can_send()` instead.

gopakumarce, 2021-03-08:
ah ok. I misunderstood .. I saw that can_send() checks for self.tx_buffer.is_full() and I just assumed that is_full is the same as is_capacity() == 0. Thanks for clarifying my misunderstanding

closed 2021-03-08 by gopakumarce