smoltcp-rs/smoltcp #438: DHCP client appears to duplicate request after an ACK

ryan-summers, 2021-03-12:
During the DHCP negotiation, the DHCP server appears to quickly respond to a DHCP REQUEST from the client. However, the client appears to ignore this DHCPACK and re-sends a DHCP REQUEST with a doubly incremented transaction ID:
[image]

The end result is still a functional DHCP client, but it appears that the client may be doing something wrong here internally.

Dirbaio, 2021-03-31:
I can't reproduce this, I'm only seeing one request/ack cycle in my hardware.

Interesting that the repeated request is exactly 2 seconds after the first one. REQUEST_TIMEOUT is set to 1 second though. Strange.

Do you have more details? Smoltcp logs would be helpful.

ryan-summers, 2021-03-31:
Yeah, this is on my TODO list to further debug, but there's some work I have to do to get logging up on our platform first. This is definitely blocked off more evidence from me.

ryan-summers, 2021-05-10:
Note: This appears to be resolved by #459 as well

closed 2021-05-28 by Dirbaio