smoltcp-rs/smoltcp #441: How to serve multiple TCP connections?

qiujiangkun, 2021-03-16:
TcpSocket implementation can only serve one client at a time. Moreover,`remote_endpoint` is not exposed as pub about ip address in TcpSocket. So how can I implement a normal web server with smoltcp?

Dirbaio, 2021-03-16:
Unlike other TCP stacks there's no "TCP listener". To accept multiple connections, you need multiple sockets. Create a socket and call listen(). Once it receives a connection, create a second socket and call .listen(), etc. Serve connected sockets concurrently with waiting for the listening socket to receive a connection.

You can get the remote endpoint with the remote_endpoint() method.

qiujiangkun, 2021-03-16:
Thanks for explaination, I know how it is going to work now.

closed 2021-03-16 by qiujiangkun