smoltcp-rs/smoltcp #443: Why does RxToken require a &mut [u8] as the argument?

qiujiangkun, 2021-03-19:
I only got &[u8] somewhere, and find that RxToken requires a &mut [u8]. I can't see any use case of &mut u8. Is it for future-proof or something else?

whitequark, 2021-03-19:
It was added in case the internals of smoltcp might want to modify the incoming packet, which is usually exclusive to the TCP/IP stack due to how networking hardware is accessed. We can probably downgrade it to `&[u8]` to support more use cases since I don't think anything in smoltcp actually used that option.

qiujiangkun, 2021-03-19:
I transmuted the reference to mut, and it works just fine in this case.

closed 2021-03-19 by qiujiangkun

Sympatron, 2021-03-19:
If it's not used mutably by smoltcp it will work currently. It's just dangerous because it may break, when smoltcp decides to mutate it some day. So it would probably be better to change it to `&[u8]` if smoltcp ended up not needing it and probably won't use it in the future.
Current implementations would still work fine with this change without any change.

I started working on a smoltcp device implementation recently too and think this would be easier to implement safely if it were not mutable.

whitequark, 2021-03-19:
> I transmuted the reference to mut, and it works just fine in this case.

I think this might be unsound.

qiujiangkun, 2021-03-19:
I'm fully aware of the unsoundness, but I have to get my job done anyway. If the code break, I can review all unsafe code then