smoltcp-rs/smoltcp #444: Efficient polling

lucaszanella, 2021-03-23:
There's little information about how much polling is needed. For example, once I do `socket.send_slice(&data[..])`, is one poll sufficient?

Currently, `socket.send_slice` will enqueue the data into the socket's transmit buffer and then call `socket_egress` on ethernet's `iface`. If I stick with the technique of `socket.send_slice` and then `iface.poll`, every time, will the data arrive at the ethernet device once a full tcp request is send? 

My idea was to poll after socket.send, and poll after ethernet receive.

I ask the same question for the polling on the ethernet side: once data arrives from ethernet, calling poll on the ethernet's `iface` will call `socket_ingress` which will call `rx_token` consume function, which will consume some bytes (possibly not all) from the ethernet. Maybe I should call poll just after ethernet receives data, and then call again until all pending data from ethernet is consumed?

Things seem to be not so easy. For example, 

```
    pub fn poll(&mut self, sockets: &mut SocketSet, timestamp: Instant) -> Result<bool> {
        let mut readiness_may_have_changed = false;
        loop {
            let processed_any = self.socket_ingress(sockets, timestamp)?;
            let emitted_any   = self.socket_egress(sockets, timestamp)?;

```

this indicates that not both pollings is done at the same time. If one gives an error, specially the first, the second is not even ran.

All the examples so simple timed polling, so that's why I ask. I'd like to not waste CPU time polling.

whitequark, 2021-03-23:
There is no guaranteed relationship between socket operations and amount of polling necessary. Smoltcp is designed to have its interfaces be polled indefinitely at intervals returned by `poll_at`/`poll_delay`.

Dirbaio, 2021-03-23:
You need to poll after "something" has happened:
- You called any method from sockets or the interface
- PHY has received packets
- PHY has new buffer space to send packets (it'd return Exhausted before, and it's now ready again)
- poll() returning an error other than Exhausted. You need to retry in this case, it may have more stuff to do.
- The time from Interface::poll_at() has expired

If you've called poll() and it has finished successfully, there's no need to call it again until some of the above happens. Doing so is guaranteed to do nothing, because the previous poll did all that was pending, and nothing has changed since.

lucaszanella, 2021-05-04:
@Dirbaio I'm looking ito this right now. As far as I understood, after calling `send_slice` for a socket, I should poll, obviously because if I try to `send_slice` again without polling, it will overwrite what I wrote without resending.

Shouldn't then exist a function to poll a specific socket instead of all? Because after doing `socket.send_slice` I should poll just `socket`, not all of them.

Dirbaio, 2021-05-04:
> if I try to send_slice again without polling, it will overwrite what I wrote without resending.

No, it enqueues more data into the socket's buffer. If it's full, it returns `Exhausted`.

> Shouldn't then exist a function to poll a specific socket instead of all? 

That'd be one way to do it, but it'd be cleaner to internallytrack the "dirty" state of sockets and poll only those automatically. This way there'd be a performance improvement without the user having to manually handle this.

Polling non-dirty sockets is very very cheap though, so I'd expect this to make little difference unless you have 100s - 1000s of sockets.

Also note that even if you do poll, it's not guaranteed that the data is sent out immediately. The connection may be slow, so the data may take more time to get drained out of the buffer. No matter what, you must always handle short writes and Exhausted errors.

lucaszanella, 2021-05-04:
> No, it enqueues more data into the socket's buffer. If it's full, it returns Exhausted.

ok so if I always poll after `send_slice`, then I can guarantee that `send_slice` will never return `Exhausted` for this socket?

Of course when I poll I mean poll until no error returns. However, there may be cases that polling after the error would not clear the error, I guess. So it's unclear how much I should repoll after a poll error. I'm afraid it could get stuck in a loop.

Dirbaio, 2021-05-05:
> ok so if I always poll after send_slice, then I can guarantee that send_slice will never return Exhausted for this socket?

No. `poll` sends data *if it can*. It won't send data if the TCP window is full, for example. This can happen if the network is slow, or the peer application is reading data slowly. This is how TCP itself works, it's nothing specific to smoltcp. 

No matter how much you poll, you have to handle short writes (`send_slice` returning a length smaller than the slice's length, which indicates only part of the data has been written) and Exhausted errors. If you don't you *will* get incorrect behavior 100% of the time, guaranteed. Again, this is how IO works, it's nothing specific to smoltcp. `std::io::Write` and unix's `write` syscall work in exactly the same way.

You need to poll-until-no-error after either of these:
- phy::Device has received packets
- you called any function on any socket
- the `poll_at` time has passed

I recommend you look at the examples to see how this is done, in particular the `phy_wait` code

lucaszanella, 2021-05-05:
@Dirbaio shouldn't we, instead of always sending back a `lib.rs::Result`, send specific `Result`s for specific function calls? I like to treat all errors, but for example, on `socket.send_slice` I have to treat all of these:

```
pub enum Error {
    /// An operation cannot proceed because a buffer is empty or full.
    Exhausted,
    /// An operation is not permitted in the current state.
    Illegal,
    /// An endpoint or address of a remote host could not be translated to a lower level address.
    /// E.g. there was no an Ethernet address corresponding to an IPv4 address in the ARP cache,
    /// or a TCP connection attempt was made to an unspecified endpoint.
    Unaddressable,

    /// The operation is finished.
    /// E.g. when reading from a TCP socket, there's no more data to read because the remote
    /// has closed the connection.
    Finished,

    /// An incoming packet could not be parsed because some of its fields were out of bounds
    /// of the received data.
    Truncated,
    /// An incoming packet had an incorrect checksum and was dropped.
    Checksum,
    /// An incoming packet could not be recognized and was dropped.
    /// E.g. an Ethernet packet with an unknown EtherType.
    Unrecognized,
    /// An incoming IP packet has been split into several IP fragments and was dropped,
    /// since IP reassembly is not supported.
    Fragmented,
    /// An incoming packet was recognized but was self-contradictory.
    /// E.g. a TCP packet with both SYN and FIN flags set.
    Malformed,
    /// An incoming packet was recognized but contradicted internal state.
    /// E.g. a TCP packet addressed to a socket that doesn't exist.
    Dropped,
}
```

but some of them could never happen in `socket_send` like `Checksum`, `Truncated`, `Finished`, etc. It gets confusing, I don't know which ones I have to consider. For `poll` I believe almost all of these could occur. For `recv`, only some of them. 

Or, at least, we should write somewhere the possible variants returned by `tcp_socket.send_slice`, `udp_socket.send_slice`, `stack.poll`, etc. 

I think that for tcp send are `Exhausted` and `Illegal`, for udp send are `Exhausted` and `Illegal`. Although I could not find how `Exhausted` is returned, since
```

    pub fn send<'b, F, R>(&'b mut self, f: F) -> Result<R>
            where F: FnOnce(&'b mut [u8]) -> (usize, R) {
        //send_impl only returns Err(Error::Illegal) https://github.com/smoltcp-rs/smoltcp/blob/09f941851ded868b759e7db067abd67f2fe136a6/src/socket/tcp.rs#L839
        self.send_impl(|tx_buffer| {
            //enqueue_many_with only returns what `f` returns: https://github.com/smoltcp-rs/smoltcp/blob/09f941851ded868b759e7db067abd67f2fe136a6/src/storage/ring_buffer.rs#L171
            tx_buffer.enqueue_many_with(f)
        })

       //f in this case returns (size, size)
    }
```

lucaszanella, 2021-05-24:
Implementing it here

https://github.com/smoltcp-rs/smoltcp/issues/474

Dirbaio, 2022-06-06:
Closing due to inactivity. If you have more questions, feel free to open new issues!

also @lattice0 re the errors: check out #617

closed 2022-06-06 by Dirbaio