smoltcp-rs/smoltcp #458: Bogus ARP requests when no IP configured

Dirbaio, 2021-04-01:
When DHCP isn't up, the interface has an address of 0.0.0.0/0 which is actually a CIDR that matches all IPs out there. If you try to connect to a WAN IP in that state it actually sends ARP requests trying to resolve it. It seems we're missing a check that treats 0.0.0.0/0 specially? Alternatively Interface could have a `[Option<Cidr>]` so we can represent "don't have an IP" as an actual None.

[image]

jordens, 2021-04-08:
0.0.0.0/**32** might be less likely to be caught up like that. Or `None` as you say. In any case it seems that if the address is unspecified (be that 0.0.0.0 or `None`), most egress should be explicitly prevented.

Dirbaio, 2023-01-15:
Fixed in #719. 

Users are no longer supposed to set `0.0.0.0/0` as an IP address, instead they should set no IP addr at all.

closed 2023-01-15 by Dirbaio