smoltcp-rs/smoltcp #467: TcpSocket can transmitt packages in wrong order

SiebenCorgie, 2021-04-28:
I observed that a tcp connection where one socket is just sending and one socket is just receiving emits a wrong package after some time.

I programmed a small app to demonstrate the problem here: https://gitlab.com/tendsinmende/smoltcp_tcp_bug
I tried the app on ArchLinux + AMD and ArchLinux + Intel. 

# What happens? 
The symptom is, that after some time the received test string ends to early and is filled with the start of the next test string instead. 
In my case the test string is
```
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Cras ac feugiat sapien, et condimentum quam. Phasellus nec faucibus nulla. Vivamus at bibendum mauris. Aliquam erat volutpat. Ut rutrum arcu sed nisi efficitur, ac sagittis lacus bibendum. Phasellus et ante lectus. Integer vestibulum, dolor consequat scelerisque fermentum, neque neque ultrices dolor, nec ultrices nisi lectus sit amet tellus. Morbi scelerisque quis mi nec facilisis. Morbi porttitor tortor rhoncus mauris blandit sodales. Etiam elementum nunc a laoreet vulputate. Curabitur interdum imperdiet viverra.
Donec elementum porta risus, in commodo libero sodales eget. Suspendisse porttitor vitae mi ut volutpat. Proin eget accumsan neque. Vestibulum quis ex placerat, semper odio in, lobortis nulla. Nulla sollicitudin eget tortor sed venenatis. Phasellus eget purus ligula. Nam fringilla ultrices dapibus. Vivamus iaculis elementum lectus id malesuada.
Nulla ultrices erat et vulputate varius. Pellentesque nec massa viverra, pharetra nulla a, rutrum 
```

However, after some time I'll receive a broken package like this one:
```
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Cras ac feugiat sapien, et condimentum quam. Phasellus nec faucibus nulla. Vivamus at bibendum mauris. Aliquam erat volutpat. Ut rutrum arcu sed
Lorem ipsum dolor sit amet, consectetur adipiscing elit. Cras ac feugiat sapien, et condimentum quam. Phasellus nec faucibus nulla. Vivamus at bibendum mauris. Aliquam erat volutpat. Ut rutrum arcu sed nisi efficitur, ac sagittis lacus bibendum. Phasellus et ante lectus. Integer vestibulum, dolor consequat scelerisque fermentum, neque neque ultrices dolor, nec ultrices nisi lectus sit amet tellus. Morbi scelerisque quis mi nec facilisis. Morbi porttitor tortor rhoncus mauris blandit sodales. Etiam elementum nunc a laoreet vulputate. Curabitur interdum imperdiet viverra.
Donec elementum porta risus, in commodo libero sodales eget. Suspendisse porttitor vitae mi ut volutpat. Proin eget accumsan neque. Vestibulum quis ex placerat, semper odio in, lobortis nulla. Nulla sollicitudin eget tortor sed venenatis. Phase
```

# Observations 
As you can see I implemented a custom device to test this. Apparently the problem already occurs in the sending socket. While the correct string is inserted into the send-queue via `socket.send_slice()`, the device appears to receive an already broken string when parsing the TCP package inside the Ethernet package that is being send over the device.

While Tcp is able to transmit the package on the device out of order (as far as I understood) there seems to be a problem when splitting the package up and putting them back together in order on the receiving end. 

The first part is send (in my example until `[..]Ut rutrum arcu sed`) but the rest of this package is never send afterwards. Instead a new package is send. 

I'm not sure if this is a duplicate of [this](https://github.com/smoltcp-rs/smoltcp/issues/452) issue, or something new. I added a print to the transmit token on this [branch](https://gitlab.com/tendsinmende/smoltcp_tcp_bug/-/tree/transmit_print). When executed the Tokens are consumed in order, until at one point it isn't in order anymore.

# What changes the behavior?
1. When increasing the buffer sizes of the packages the bug occurs later. 
2. Sometimes in debug builds a `attempt to multiply with overflow` occurs here: https://github.com/smoltcp-rs/smoltcp/blob/v0.7.x/src/socket/tcp.rs#L143

# Context
OS: ArchLinux 
rustc: rustc 1.52.0-nightly (8e54a2113 2021-02-13)
smoltcp: 0.7.1
occurs in debug and release build

Is my expectation wrong that the data returned by `socket.recv()` is in the same order as given to `socket.send()`?

Any Ideas if I'm missing something or where to look for the bug?

Cheers 
Tendsin Mende

Dirbaio, 2021-04-28:
The bug is in this snippet. `send_slice` may enqueue only part of the slice you provide, which can cause the messages to be "cut" like you're seeing.

```rust
                    let size = socket.send_slice(&TESTBUFFER).unwrap();
                    //println!("Send {}", size);
                    if size == 0{
                        //println!("NOT SEND");
                    }else{
                        //println!("SEND");
                    }
```

SiebenCorgie, 2021-04-28:
Hi, thanks for the fast reply! 

I added tracking of which parts of the buffer where sent to the example, but it seems to persist on my side.
According to the `println` the `send_slice()` function enqueues the whole package, when enqueuing anything at least.

Dirbaio, 2021-04-28:
There's another bug in the receiving code: you're setting `ptr` to 0 in each poll iteration, throwing away data. On one iteration you may receive only part of a message, you have to keep that data for when the rest arrives later.

https://gitlab.com/tendsinmende/smoltcp_tcp_bug/-/blob/master/src/main.rs#L216

SiebenCorgie, 2021-04-28:
That fixed it indeed, thank you for your time!

closed 2021-04-28 by SiebenCorgie