smoltcp-rs/smoltcp #472: Version 0.7.1 does not follow semantic versioning rules with 0.7.0

ryan-summers, 2021-05-10:
It appears that the 0.7.1 release does not follow semantic versioning rules with 0.7.0 - specifically, if the smoltcp dependency is specified as:
```
[dependencies.smoltcp]
version = "0.7.0"
features = ["ethernet", "proto-ipv4"]
```

Cargo may instead choose to use smoltcp =0.7.1. If the above feature flags are provided to smoltcp 0.7.1, the build will fail.

Given that the feature flags changed between 0.7.0 and 0.7.1, it seems that 0.7.1 is not following the semantic versioning rules. 

I would recommend that we yank 0.7.1 and re-release it as 0.8.0

cc: @Dirbaio

ryan-summers, 2021-05-10:
The key here is that the `stm32h7xx-hal` uses smoltcp as a dependency, but only implements `phy::Device` for the ethernet module so that external users can continue to use the PHY as they would like.

Whether this is a valid means of using smoltcp is up for debate, but it _did_ compile, and since the stm32h7xx-hal crate doesn't actually do any protocol parsing (it just implements the trait), this was a valid configuration.

Applications then also specified a `smoltcp` dependency that then had to have all the necessary feature flags.

Maybe a good solution here is to move the phy::Device trait to some external crate so that HALs can implement it without pulling in smoltcp?

ryan-summers, 2021-05-10:
Hmm.. Okay, after investigating further, it appears that there's a bug in the `stm32h7xx-hal` Cargo.toml that causes the feature-flags in dev-dependencies to propagate into the original `smoltcp` dependency, which enabled `sockets-raw`. Let me see if I can resolve things outside of smoltcp and I'll see if this all resolves.

ryan-summers, 2021-05-10:
Closing this as resolved - it looks like dev-dependency unification bugs in Cargo caused this issue in `stm32h7xx-hal`. Thanks for the feedback!

closed 2021-05-10 by ryan-summers

Sympatron, 2021-05-10:
@ryan-summers Maybe using [the new feature resolver](https://doc.rust-lang.org/beta/cargo/reference/resolver.html#resolver-versions) will solve this?

ryan-summers, 2021-05-10:
Yeah, but it's still in beta, unfortunately, so we can't use that in the `stm32h7xx-hal`

Sympatron, 2021-05-10:
It got stabilized recently: https://github.com/rust-lang/cargo/blob/master/CHANGELOG.md#cargo-151-2021-03-25

ryan-summers, 2021-05-10:
Ooh, I was wondering. I'll look into it - thanks for the intel