smoltcp-rs/smoltcp #473: Embedded question: Bypassing smoltcp's Ethernet layer?

john-terrell, 2021-05-15:
This is a question and not an issue per-se: The embedded hardware on my Microchip board generates its own Ethernet frames.    Does smoltcp have any mechanism to bypass smoltcp's own Ethernet frame generation and supply IP frames directly to the Device object?

Thanks!

Dirbaio, 2021-05-15:
In smoltcp `master` (not yet released in crates.io) you can specify the "medium" of a Device. You can pick the IP medium, in which case your Device will send/receive IP packets, not Ethernet frames. If the hardware generates the Ethernet frames and does neighbor discovery/ARP etc on its own it should work.

Which chip/board is it?

john-terrell, 2021-05-18:
Thanks for the info.   I'm implementing the ethernet driver for the atsam-rs project.    https://github.com/atsam-rs

Follow-on question:  It looks like enabling proto-dhcpv4 also forcibly enables medium-ethernet.   Is that due to needing to get to the underlying MAC address?

john-terrell, 2021-05-22:
Thanks for the guidance.

closed 2021-05-22 by john-terrell