smoltcp-rs/smoltcp #474: Async smoltcp effort

lucaszanella, 2021-05-24:
I'm starting an effort to make polling invisible for the user and implement tokio `AsyncWrite` and `AsyncRead` into smoltcp, so it can be used with libraries that use tokio for tcp. It's a new crate with smoltcp as a dependency, not a modification of smoltcp.

https://github.com/lucaszanella/async_smoltcp/blob/master/src/async_smoltcp.rs

It's in the very early stages and it's based on unmerged work from @Dirbaio (that adds ethernet support). When it merges I'll make it work here. (are there estimates on when it will be merged?)

Since the smoltcp project is low level in relation to polling because it's intended for microcontrollers, I'm bringing smoltcp for general usage. The idea is to be able to share a stack with multiple threads and not worry about polling or anything, just pass the socket for something that expects `AsyncRead` and `AsyncWrite`. I'm using dynamically allocated buffers because I need to share the stack with multiple threads anyway so references would not work. 

I also made an effort to make one single struct to support tap, tun and virtual tun.

It's not working yet but I expect it to work soon. I'll add an example using hypervpn that I already made work.

**It'd be nice if someone could help, there are lots of things to do specially with preventing memory copy on socket data arrival and also pick the best polling method**

qiujiangkun, 2021-06-07:
I'm doing my own async async wrapper, with a bit of challanges.
The biggest problem is `SocketHandle`. I need to own at least `Arc<TcpChannel>`(instead of the whole `SocketSet` and `SocketHandle`) and read_buffer, write_buffer. However, current model does not allow me to do that, so I have to change the source code to acheive have `Arc<TcpChannel>`. 
Ideally I prefer shared memory for states and buffers in `TcpChannel`, so unsafe is involved.
I think the cause is the goal behind smoltcp: simple and without heap. When we have heal alloc, we can do much better.

spacemeowx2, 2021-06-13:
Wow, so many people are doing the same thing! Including me.

Here is my version of async smoltcp.

https://github.com/spacemeowx2/tokio-smoltcp

You can try with the `pcap` example:

```sh
cargo build --example pcap && sudo ./target/debug/examples/pcap -h
```

lucaszanella, 2021-06-13:
@qiujiangkun forgot to answer, sorry. Did you finish your version by modifying the source code of `smoltcp`? I'd prefer not to modify it but I'd like to take a look at yours. Maybe the modifications are good enough that they could fit in `smoltcp`.

@spacemeowx2 I've just took a superfical read of yours. It looks like you have a thread that polls the stack (actually the `run` function is `async`, I don't know why a thread isn't used). Anyways, it looks like this 'thread' and the sockets share the `sockets` set. That is, they both fight over the `sockets` lock. I thought of something like this but my main concern was that each socket must be polled every time something happens (in/out of data, etc). That is, when some data is received/sent over the wire, all the sockets should be polled immediately. I think it's possible that there's a situation where data is received in some socket and its `AsyncRead` does not call it fast enough, making the next data that arrives in the socket overwrite what was already there.

Did you think of this while designing your solution? Do you think I'm right?

On my approach I'm making the stack poll every time it's needed. That is, at every operation, all sockets are polled. When a socket has data, it immediately sends this data over a channel to the socket on the other side, so if more data arrives, there's no problem of it being overriden. I'm not happy about this solution because it does an extra copy before the data arrives on the `AsyncRead` but I couldn't think of anything better. I see that your solution immediately copies from the rx buffer to the `AsyncRead`'s `buf`, which is better in terms of not doing unnecessary copies.

spacemeowx2, 2021-06-13:
@lucaszanella 

> I don't know why a thread isn't used

Because I want to `async` as much as possible, the raw socket could also be `async`:

https://github.com/spacemeowx2/tokio-smoltcp/blob/ad7f043a07841961d2b9ff9bb3785e7111b220fc/src/util/unix.rs#L22

> making the next data that arrives in the socket overwrite what was already there

This can't happen. `TcpSocket` has an internal buffer. If the buffer is full, the underlying protocol(TCP) will not send an ACK to the other side, so the other side will not send more data than the buffer size. So overwriting will not happen.

qiujiangkun, 2021-06-13:
> @qiujiangkun forgot to answer, sorry. Did you finish your version by modifying the source code of smoltcp? I'd prefer not to modify it but I'd like to take a look at yours. Maybe the modifications are good enough that they could fit in smoltcp.

Yes I have finished my version. It barely works but the speed is depressing(even with only smoltcp). I used a channel to communicate with smoltcp, which works fine. 
```rust

#[derive(Clone)]
pub struct TcpChannelBuilder {
    stub: SmoltcpController,
    pub addr: String,
    pub port: u16,
}

impl Debug for TcpChannelBuilder {
    fn fmt(&self, f: &mut Formatter<'_>) -> std::fmt::Result {
        f.debug_struct("TcpBuilder")
            .field("addr", &self.addr)
            .field("port", &self.port)
            .finish_non_exhaustive()
    }
}

impl TcpChannelBuilder {
    pub fn connect_to(addr: &str, port: u16) -> Self {
        let stub = get_global_stub();
        Self::with_controller(stub, addr, port)
    }
    pub fn new(addr: &str, port: u16) -> Self {
        let stub = get_global_stub();
        Self::with_controller(stub, addr, port)
    }
    pub fn with_controller(stub: SmoltcpController, addr: &str, port: u16) -> Self {
        Self {
            stub,
            addr: addr.to_owned(),
            port,
        }
    }
}

impl AsyncChannelBuilder for TcpChannelBuilder {
    type Channel = TcpChannel;

    fn connect(&self) -> BoxFuture<'static, Result<Self::Channel>> {
        let addr = self.addr.to_owned();
        let port = self.port;
        let stub = self.stub.clone();
        let command = SmoltcpTcpRequest::Connect(self.addr.to_owned(), self.port);
        Box::pin(async move {
            let channel = stub
                .send_wait_command(command.into())
                .await
                .into_tcp_channel_result()?;
            let local_port = channel.get_socket_mut().local_endpoint().port;
            let timeout = std::time::Instant::now();
            loop {
                if std::time::Instant::now() - timeout > std::time::Duration::from_secs(5) {
                    break Err(std::io::Error::new(
                        ErrorKind::TimedOut,
                        format!("Could not connect to {}:{} in time", addr, port),
                    ));
                }
                match channel.get_socket_mut().state() {
                    TcpState::Closed => {
                        break Err(std::io::Error::new(
                            ErrorKind::ConnectionAborted,
                            format!(
                                "Could not connect from {} to {}:{}. state: {}",
                                local_port,
                                addr,
                                port,
[... 126 lines cut ...]
        map_non_block(self.read(buf), cx)
    }
}

impl AsyncWrite for TcpChannel {
    fn poll_write(
        mut self: Pin<&mut Self>,
        cx: &mut Context<'_>,
        buf: &[u8],
    ) -> Poll<Result<usize>> {
        map_non_block(self.write(buf), cx)
    }

    fn poll_flush(self: Pin<&mut Self>, _cx: &mut Context<'_>) -> Poll<Result<()>> {
        Poll::Ready(Ok(()))
    }

    fn poll_close(self: Pin<&mut Self>, _cx: &mut Context<'_>) -> Poll<Result<()>> {
        self.get_socket_mut().close();
        Poll::Ready(Ok(()))
    }
}

impl Unpin for TcpChannel {}
```

lucaszanella, 2021-06-14:
@spacemeowx2 I understand what you said but what if the rx buffer is smaller than an entire tcp packet? The tcp packet would arrive in chunks so it should be consumed lots of times before an ACK is sent, I guess. I'm not an expert in smoltcp internals. For example, it looks like it's here that a packet is written to `rx_buffer`:

https://github.com/smoltcp-rs/smoltcp/blob/e4241510337e095b9d21136c5f58b2eaa1b78479/src/socket/tcp.rs#L1564

I don't see any logic that makes the packet be held in case `rx_buffer` is full. I think that your logic works if `rx_buffer` is bigger than an entire tcp packet. This way, every time it's filled, it's going to be with an entire tcp packet. I'm also assuming that the ACK is only sent after the data is taken off of the socket, I don't see where it happens, I'm assuming since you said. Also, what about UDP packets or RAW packets? They would be overriden some times since they have no ACK.

qiujiangkun, 2021-06-17:
Finally, my version works normally. With dpdk device, it even reading faster several times than linux's TCP stack. I use channels to make an local RPC to communicate with smoltcp. I use `Vec<u8>` channels to pass tcp data around, which is not optimal but at least thread-safe.

lucaszanella, 2021-06-18:
@qiujiangkun can I see it? Does it need source code modification on smoltcp?

qiujiangkun, 2021-06-18:
It's in a private repo so I'm afraid not. It doesn't need modifications on smoltcp

lucaszanella, 2021-06-20:
https://github.com/lucaszanella/async_smoltcp

works but polling is inneficient (big inneficient loop for now). It works with my OpenVPN client. Still gotta test tun/tap but it should work. 

Gonna work on polling by making it poll onyl when necessary, then gonna fix error handling, currently it has some unwraps.

Dirbaio, 2022-06-06:
Closing due to inactivity. If you have more questions, feel free to open new issues!

closed 2022-06-06 by Dirbaio