smoltcp-rs/smoltcp #489: panic: attempt to subtract sequence numbers with underflow

spacemeowx2, 2021-06-10:
I got panic when using in real-world application (using [tokio-smoltcp](https://github.com/spacemeowx2/tokio-smoltcp) without unsafe code).

https://github.com/smoltcp-rs/smoltcp/blob/216a94117b0a139f119ae780e3124548e187c3b9/src/wire/tcp.rs#L56

https://github.com/smoltcp-rs/smoltcp/blob/216a94117b0a139f119ae780e3124548e187c3b9/src/socket/tcp.rs#L1794

I am not very familiar with Tcp. So I debugged to get the TCP state when panic.

```Rust
self.local_seq_no = 1;
self.remote_win_len = 64;
self.remote_last_seq = 518;
self.state = CloseWait;
```

Please tell me what I need to provide when panic.

Dirbaio, 2021-06-11:
Thanks for the report @spacemeowx2 !

#490 fixes one case where this could happen, which is when the remote side annoinces a certain window size and then later shrinks it on its own (not in response to us sending data). I find this quite surprising, since it's strongly discouraged and I've never seen that in the wild (which is why that bug had gone unnoticed). 

Could you provide the following?
- A packet capture (.pcap, .pcapng) of the traffic that causes the bug.
- What OS / network stack is the remote device you're talking with? 

This will allow checking if it's really the remote side shrinking its window, or some other edge case that also needs fixing!

Thanks! :)

spacemeowx2, 2021-06-11:
Thank you for quick response!

I can't tell what OS is the remote device because the panic happened when I surfing the Internet.

I tried to reproduce this issue but found another panic:

```
panicked at 'attempt to subtract with overflow'
```

https://github.com/smoltcp-rs/smoltcp/blob/216a94117b0a139f119ae780e3124548e187c3b9/src/socket/tcp.rs#L1267

1267.pcapng:

[1267.zip](https://github.com/smoltcp-rs/smoltcp/files/6639925/1267.zip)

Dirbaio, 2021-06-11:
>happened when I surfing the Internet.

Browsing the Internet with smoltcp? how do you do that? Sounds fun, seems to be a tough torture test :) 

> found another panic:

Wow, just wow. That one seems a server TCP bug...?? When you send `SYN seq=N`, the server should reply with `SYNACK ack=N+1`. That particular IP responds to `SYN seq=0` with `SYNACK ack=0`. It responds fine to SYNs with nonzero seq. Normal TCP clients work fine because initial seq is randomized, while smoltcp always starts with seq=0. It also fails to randomize its own starting seq :rofl: 

Can be reproduced with:
```
[dirbaio@mars ~]$ sudo nping -vv -c 1 --tcp -p 443 --flags syn --seq 0 76.223.126.88
SENT (0.0452s) TCP [192.168.1.34:62978 > 76.223.126.88:443 S seq=0 ack=0 off=5 res=0 win=1480 csum=0x255B urp=0] IP [ver=4 ihl=5 tos=0x00 iplen=40 id=2907 foff=0 ttl=64 proto=6 csum=0xe273]
RCVD (0.0566s) TCP [76.223.126.88:443 > 192.168.1.34:62978 SA seq=0 ack=0 off=6 res=0 win=65535 csum=0x190B urp=0 <mss 0>] IP [ver=4 ihl=5 tos=0x00 iplen=44 id=2907 foff=0 ttl=246 proto=6 csum=0x2c6f]
                                                              ^^^^^^^^^^^
                                                                 WRONG

[dirbaio@mars ~]$ sudo nping -vv -c 1 --tcp -p 443 --flags syn --seq 1 76.223.126.88
SENT (0.0351s) TCP [192.168.1.34:48093 > 76.223.126.88:443 S seq=1 ack=0 off=5 res=0 win=1480 csum=0x5F7F urp=0] IP [ver=4 ihl=5 tos=0x00 iplen=40 id=47756 foff=0 ttl=64 proto=6 csum=0x3342]
RCVD (0.0467s) TCP [76.223.126.88:443 > 192.168.1.34:48093 SA seq=2630440566 ack=2 off=6 res=0 win=65535 csum=0x5A3A urp=0 <mss 1460>] IP [ver=4 ihl=5 tos=0x00 iplen=44 id=0 flg=D foff=0 ttl=245 proto=6 csum=0xf8c9]
                                                              ^^^^^^^^^^^^^^^^^^^^
                                                                 OK
```

Of course smoltcp shouldn't crash in this case, I'll fix it. 

The fix will ignore the packet however (because it's incorrect), so communication will still not work. I'm not sure what the best fix is, probably start with seq=1000 or whatever instead of seq=0. Or implement proper randomization.

closed 2021-06-11 by Dirbaio

spacemeowx2, 2021-06-11:
Thank you for your tough work! 

> how do you do that?

You can try the `rabbit-digger-pro` that I am developing if you are intrested :)

```sh
git clone --recurse-submodules git@github.com:rabbit-digger/rabbit-digger-pro.git
cd rabbit-digger-pro
# Modify and paste `config.yaml` below. Making sure you have dependences install require by `pcap` crate.
# https://crates.io/crates/pcap
./cgr
# maybe you need to run as root
# sudo ./target/debug/rabbit-digger-pro
```

`config.yaml`
```yaml
net:
  raw:
    type: raw
    device: "eth0"
    mtu: 1400
    # must be the same with your adapter if you are using wireless adapter
    ethernet_addr: 70:85:C2:71:43:1D
    # pick a free ip in your subnet
    ip_addr: 192.168.233.4/24
    # your gateway
    gateway: 192.168.233.1
  # DNS resolving is not implemented in `raw` yet. So use local domain resolver here.
  resolve_raw:
    type: resolve
    net: raw
    ipv6: false
server:
  mixed:
    # could also be `http` or `socks5`. This `type` supports both `http` and `socks5`.
    type: http+socks5
    # the proxy server is now on 11221. Then you can use smoltcp to browser the Internet by this proxy.
    bind: 127.0.0.1:11221
    net: resolve_raw
```

Dirbaio, 2021-06-11:
Really cool! will definitely try it out.

Fixes released in v0.7.4

PR #490 (2021-06-11): Fix "subtract sequence numbers with underflow" on remote window shrink.
Fixes #489
commit e798dfdd (2021-06-11): Fix "subtract sequence numbers with underflow" on remote window shrink.

Fixes #489
PR #492 (2021-06-11): tcp: use nonzero initial sequence number.
Fixes #489
commit 5722691c (2021-06-11): Fix "subtract sequence numbers with underflow" on remote window shrink.

Fixes #489