smoltcp-rs/smoltcp #493: Real world tcp read benchmark never finishes.

qiujiangkun, 2021-06-13:


spacemeowx2, 2021-06-13:
Looks like your code forgot to call `phy_wait`. So the poll loop actually spins waiting for RawSocket to have data which is very inefficient.

See example:

https://github.com/smoltcp-rs/smoltcp/blob/e4241510337e095b9d21136c5f58b2eaa1b78479/examples/httpclient.rs#L110

https://github.com/smoltcp-rs/smoltcp/blob/e4241510337e095b9d21136c5f58b2eaa1b78479/src/phy/sys/mod.rs#L26

qiujiangkun, 2021-06-13:
Thanks for pointing out. Though I do let it spin on purpose, since performance is my first priority at any cost.

spacemeowx2, 2021-06-13:
I think the data consumer is not quick enough. replace `if tcp.may_recv()` to `while tcp.can_recv()` may help.

```rust
        let mut data = [0u8; 1024 * 8]; 
        if tcp.may_recv() {
            tcp.recv_slice(&mut data)?;
        }
```

And I must say [`write`](https://doc.rust-lang.org/std/io/trait.Write.html#tymethod.write) doesn't guarantee all data will be sent. Use [`write_all`](https://doc.rust-lang.org/std/io/trait.Write.html#method.write_all) instead.

qiujiangkun, 2021-06-13:
```rust
while tcp.may_recv() {
     tcp.recv_slice(&mut data)?;
}
```
doesn't help much. In the end smoltcp seems stuck waiting for something, so the bottleneck is not CPU

qiujiangkun, 2021-06-16:
I have spotted serveral severe issues.

1. MTU on localhost is not honored when the remote machine is trying to detect the proper MTU. This is very significant in AWS. My network interface for smoltcp only supports 1500, and that on the remote machine supports 9000. Then the remote machine supposes MTU on path to be 3000 since smoltcp did not send out an icmp message `Fragmentation Needed` while receiving a 2900 bytes sized packet. It simply get a checksum error and drops.

2. While receiving a checksum errored TCP packet, smoltcp did not send out ACK requiring the remote to retransmit, but instead replying on the remote to time out.

pothos, 2021-06-16:
Can you explain the second point a bit more?

Besides the issues you mentioned, these two points from the [TCP README section](https://github.com/smoltcp-rs/smoltcp#tcp-layer) are relevant when you have stuck connections:

- Probing Zero Windows is **not** implemented.
- Packetization Layer Path MTU Discovery [PLPMTU](https://tools.ietf.org/rfc/rfc4821.txt) is **not** implemented.

However, I guess most remote machines don't have PLMTU activated either and this won't help here. Your remote machine probably needs some other trick to know the right path MTU if smoltcp is not to blame for not sending an ICMP out for this large packets.

qiujiangkun, 2021-06-16:
I'm running `tracepath` on the remote machine.
```
 tracepath some.ip.v4.address                                                                   

 1?: [LOCALHOST]                      pmtu 9001
 1:  no reply
 2:  no reply
 3:  no reply
 4:  no reply
 5:  no reply
 ...
```
And I'm only getting the following repeatedly.
```log
EthernetII src=06-db-18-9e-7d-66 dst=06-67-ae-9e-c3-0a type=IPv4
\ (truncated packet)    
Jun 16 09:22:01.745 TRACE main smoltcp_raw_tcp_perf: localhost <- EthernetII src=06-db-18-9e-7d-66 dst=06-67-ae-9e-c3-0a type=IPv4
   \ (truncated packet)

```
I suppose the remote linux machine are trying to detect the smoltcp side's MTU but couldn't get one. So it is trying to send extra large frames based on its MTU 9001.

> Packetization Layer Path MTU Discovery PLPMTU is not implemented

I don't need smoltcp's PLPMTU, but I need smoltcp to response to such extra large ethernet frames with ICMP message.

qiujiangkun, 2021-06-16:
> Your remote machine probably needs some other trick to know the right path MTU if smoltcp is not to blame for not sending an ICMP out for this large packets.

Indeed, my network interface has trouble supporting larger frames. But the remote machine may not be something I can control, so I need to send out an ICMP message in smoltcp.

closed 2021-06-17 by qiujiangkun