smoltcp-rs/smoltcp #497: RawSocket buffer size is too small

qiujiangkun, 2021-06-16:


spacemeowx2, 2021-06-16:
As far as I know there is no such concept as remote MTU.

qiujiangkun, 2021-06-16:
By remote MTU, I mean the MTU from the view of the other endpoint.

spacemeowx2, 2021-06-16:
If you get wrong MTU, blame this:

https://github.com/smoltcp-rs/smoltcp/blob/027f255f904b9b7c4226cfd8b2d31f272ffa5105/src/phy/sys/raw_socket.rs#L36

qiujiangkun, 2021-06-16:
The MTU is correct, but MTU should not be the maximum buffer size

spacemeowx2, 2021-06-16:
What's the `remote MTU` value and `local MTU` here?

The correct MTU should be the maximum value of them.

I think you are resolving the Y of [XY problem](https://en.wikipedia.org/wiki/XY_problem).

qiujiangkun, 2021-06-16:
remote MTU is 9000
local MTU is 1500.

local smoltcp is trying to receive a 9000 byte sized frame with a 1500 byte sized buffer.

spacemeowx2, 2021-06-16:
Ahh I see the value here

https://github.com/smoltcp-rs/smoltcp/issues/493#issuecomment-862031207

And this it the X needed to be solved.

https://github.com/smoltcp-rs/smoltcp/issues/493#issuecomment-862237564

If your interface does support receive 9000 bytes or 3000bytes, the MTU shouldn't be 1500.

If not, increasing the receive buffer doesn't help.

qiujiangkun, 2021-06-16:
> In the context of Internet Protocol, MTU refers to the maximum size of an IP packet that can be transmitted without fragmentation over a given medium. 

https://www.wikiwand.com/en/Maximum_transmission_unit

MTU means the maximum amount that can be transmitted, not received. Adjusting local MTU could accidentally fix it, but in the wrong way. Moreover, the `local` network card does not support larger MTU size, and I can't simply upscale its MTU.

Dirbaio, 2021-06-16:
As far as I know, MTU in Linux means the maximum IP packet size for both sending and receiving. If you set your MTU to 1500, you will not receive IP packets bigger than 1500 (ie ethernet frames bigger than 1514). 

So you shouldn't be receiving big packets if your MTU is 1500.

This may be caused by offloads, probably GRO. Check which is enabled with `ethtool --show-offload eth0`. Try disabling GRO with `ethtool --offload eth0 generic-receive-offload off`. If it still doesn't fix it, try disabling other offloads.

qiujiangkun, 2021-06-16:
> As far as I know, MTU in Linux means the maximum IP packet size for both sending and receiving. If you set your MTU to 1500, you will not receive IP packets bigger than 1500 (ie ethernet frames bigger than 1514).
> So you shouldn't be receiving big packets if your MTU is 1500.

No, I should not receive IP packets bitter than 1500, unless I am working with raw sockets.

> This may be caused by offloads, probably GRO. Check which is enabled with ethtool --show-offload eth0. Try disabling GRO with ethtool --offload eth0 generic-receive-offload off. If it still doesn't fix it, try disabling other offloads.

It turns out to be `tcp-segmentation-offload`. The command should be `ethtool --offload eth0 tso off` on Ubuntu. After manually setting sender's MTU to 1500 and disabling TSO, the throughput goes normal(at least on my server).

Still, if the server does not disable `tcp-segmentation-offload` or set a larger MTU, smoltcp over RawSocket would not work.

Dirbaio, 2021-06-16:
Are you using TCP? TCP exchanges the MSS in SYN/SYNACK packets so that both ends know the max MTU. You shouldn't see big packets even if the other side has a bigger MTU configured.

Yes, this is a known issue with RawSocket. IPv4 reassembly can also cause "packets bigger than MTU" issues. RawSocket is quite a hack because it captures packets destined for the host, so the Linux network stack processes packets assuming they're only for its own consumption, and does evil things to them.

You may want to use TUN/TAP interfaces with routing/bridging instead, they don't have all these issues.

Dirbaio, 2021-06-16:
As for how to fix this in smoltcp, we can make RawSocket print a warning if it receives packets bigger than MTU. I'm not sure if much else can be done, since Device impls are not allowed to return packets bigger than the MTU, so just fixing it in RawSocket will cause issues higher in the stack.

qiujiangkun, 2021-06-16:
> Are you using TCP? TCP exchanges the MSS in SYN/SYNACK packets so that both ends know the max MTU. You shouldn't see big packets even if the other side has a bigger MTU configured.

Yes, I'm using TCP. I think the sender's `tcp-segmentation-offload` did not respect MTU for some reason(I got big packets on the sender machine). 

Now I see the differences between RawSocket and TAP/TUN. 

> As for how to fix this in smoltcp, we can make RawSocket print a warning if it receives packets bigger than MTU. I'm not sure if much else can be done, since Device impls are not allowed to return packets bigger than the MTU, so just fixing it in RawSocket will cause issues higher in the stack.

In `interface.rs`, it will give a `Error::Truncated` when the packets are larger than MTU. I tried to set the buffer size to 10000, and it did not cause any troubles.

I'm still trying to do some more testing:

- connect to a real server I was working with, which I don't have permissions to modify
- switch to dpdk device and see the performance(so that any weird problems with RawSocket is gone)

Dirbaio, 2021-06-16:
> I think the sender's tcp-segmentation-offload did not respect MTU for some reason(I got big packets on the sender machine).

with TSO enabled, Linux will produce TCP packets bigger than the MTU, hand them to the network hardware which chops them up into smaller packets before sending them on the wire. If you sniff you'll see big packets, but they'll be small on the wire and the receiving machine will receive small packets.

GRO is the opposite: the network driver receives many small packets and combines them into a bigger-than-MTU packet before passing it to Linux's network stack. You'll see big packets when sniffing or using RawSocket, but they were actually small on the wire.

Offloads are super confusing because what you see when sniffing is not what is actually on the wire.

qiujiangkun, 2021-06-16:
> with TSO enabled, Linux will produce TCP packets bigger than the MTU, hand them to the network hardware which chops them up into smaller packets before sending them on the wire. If you sniff you'll [...]

> GRO is the opposite: the network driver receives many small packets and combines them into a bigger-than-MTU packet before passing it to Linux's network stack. You'll see big packets when sniffing or using RawSocket, but they were actually small on the wire.

Thanks. If I disable TSO on sender side or GRO on receiver side, it will work.

qiujiangkun, 2021-06-17:
I have done serveral benchmarks, here's the results.

dpdk default-features 6782.008651957823 KiB/s
dpdk default-features without `socket-raw` 39231.51261627999 KiB/s
[flamegraph.zip](https://github.com/smoltcp-rs/smoltcp/files/6667951/flamegraph.zip)

raw-socket default-features  543.1268610999218 KiB/s
raw-socket default-features without `socket-raw`  1018.2408903056562 KiB/s
netcat 6587.746492850176 KiB/s

It should be stated that `socket-raw` is very inefficient somwhere in `README.md`

pothos, 2021-06-17:
You can try macvtap in passthru or bridge mode (https://github.com/ANLAB-KAIST/usnet_devices) which should be more efficient but still you'll have one syscall per packet unless a different API like io_uring is introduced.
I'm not sure if it is affected by the same GRO problem, though.

qiujiangkun, 2021-06-17:
The issue has been solved. Thank you all!

closed 2021-06-17 by qiujiangkun