smoltcp-rs/smoltcp #503: Retransmission bug?

lucaszanella, 2021-06-19:
I'm testing the latest merge from @Dirbaio  that brings Medium::Ethernet/IP support. Code that before worked before on the sketch of the now merged code isn't working. now

A simple tcp connect isn't working. Even though the tcp server on the other side responds with SYN + ACK, smoltcp decides to retransmit the connection packet again (SYN). 

```
---------------------------------------------------------------
[2021-06-19T23:33:28Z DEBUG async_smoltcp::async_smoltcp] >>>>>>> on_virtual_tun_write for buffer with len 52
[2021-06-19T23:33:28Z DEBUG async_smoltcp::async_smoltcp] ip: Some(Version4(Ipv4Header { ihl: 5, differentiated_services_code_point: 0, explicit_congestion_notification: 0, payload_len: 32, identification: 0, dont_fragment: true, more_fragments: false, fragments_offset: 0, time_to_live: 64, protocol: 6, header_checksum: 59874, source: [192, 168, 69, 2], destination: [54, 207, 20, 104], options: [] }))
[2021-06-19T23:33:28Z DEBUG async_smoltcp::async_smoltcp] transport: Some(Tcp(TcpHeader { source_port: 47675, destination_port: 80, sequence_number: 42, acknowledgment_number: 0, data_offset: 8, ns: false, fin: false, syn: true, rst: false, psh: false, ack: false, urg: false, ece: false, cwr: false, window_size: 65000, checksum: 27031, urgent_pointer: 0 }))
[2021-06-19T23:33:28Z DEBUG async_smoltcp::async_smoltcp] payload: ""
---------------------------------------------------------------
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] <<<<<<< on_virtual_tun_read for buffer with len 52
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] ip: Some(Version4(Ipv4Header { ihl: 5, differentiated_services_code_point: 0, explicit_congestion_notification: 0, payload_len: 32, identification: 0, dont_fragment: true, more_fragments: false, fragments_offset: 0, time_to_live: 240, protocol: 6, header_checksum: 14819, source: [54, 207, 20, 104], destination: [192, 168, 69, 1], options: [] }))
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] transport: Some(Tcp(TcpHeader { source_port: 80, destination_port: 47675, sequence_number: 3339187062, acknowledgment_number: 43, data_offset: 8, ns: false, fin: false, syn: true, rst: false, psh: false, ack: true, urg: false, ece: false, cwr: false, window_size: 26883, checksum: 16459, urgent_pointer: 0 }))
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] payload: ""
---------------------------------------------------------------
[2021-06-19T23:33:29Z DEBUG smoltcp::socket::tcp] #0:0.0.0.0:47675:54.207.20.104:80: retransmitting at t+0.709s
---------------------------------------------------------------
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] >>>>>>> on_virtual_tun_write for buffer with len 52
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] ip: Some(Version4(Ipv4Header { ihl: 5, differentiated_services_code_point: 0, explicit_congestion_notification: 0, payload_len: 32, identification: 0, dont_fragment: true, more_fragments: false, fragments_offset: 0, time_to_live: 64, protocol: 6, header_checksum: 59874, source: [192, 168, 69, 2], destination: [54, 207, 20, 104], options: [] }))
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] transport: Some(Tcp(TcpHeader { source_port: 47675, destination_port: 80, sequence_number: 42, acknowledgment_number: 0, data_offset: 8, ns: false, fin: false, syn: true, rst: false, psh: false, ack: false, urg: false, ece: false, cwr: false, window_size: 65000, checksum: 27031, urgent_pointer: 0 }))
[2021-06-19T23:33:29Z DEBUG async_smoltcp::async_smoltcp] payload: ""
```

You can see at the line

`[2021-06-19T23:33:29Z DEBUG smoltcp::socket::tcp] #0:0.0.0.0:47675:54.207.20.104:80: retransmitting at t+0.709s`

It tires to retransmit very fast. I tried `socket.set_timeout(Some(smoltcp::time::Duration::from_millis(10000)));` but even with this, it still retransmit in less than 1 second. 

Is `master` a work in progress? I tried the latest release but it does not have the `TunTapInterface` and etc.

Dirbaio, 2021-06-20:
TCP connecting should definitely work, even in master. Can you post the full `trace`-level logs?

lucaszanella, 2021-06-20:
@Dirbaio it looks like it was a problem on my side but anyways the `set_timeout` looks like it has no effect. It tried to retransmit after 0.7s, don't have an idea why. At least for tcp connect it does this.

closed 2021-06-20 by lucaszanella

Dirbaio, 2021-06-20:
`set_timeout` sets the time after which the socket will time out (close), not the retransmit timeout. That one is automatically set based on the RTT estimation, and that estimation starts at the ~0.7s you're seeing.