smoltcp-rs/smoltcp #507: Why do we use `waker.wake()`, instead of `waker.wake_by_ref()`?

qiujiangkun, 2021-06-27:
It is very inconvenent to set waker every time it is woken, not to mention the cost of drop and clone. These can all be avoidable easily by `wake_by_ref()`

Dirbaio, 2021-06-27:
Usually when tasks poll some future, they expect it to be woken up from it just once. If they're still interested in the result of that operation, they'll poll again which will register the waker again.

Storing the waker permanently can be inefficient if a task polls a socket once, and then is no longer interested in it. The waker will stay registered, which will wake the task unnecessarily every time something happens to the socket, again, forever.

I believe tokio/mio internally do the same for std IO. In the end a waker gets cloned and dropped for every IO operation.

It's true cloning and dropping the wakers has some cost. Note that in most no-std executors this doesn't apply, as they don't refcount wakers (tasks are statically allocated), so clone/drop of these wakers is effectively "free".

qiujiangkun, 2021-06-27:
> Storing the waker permanently can be inefficient if a task polls a socket once, and then is no longer interested in it. The waker will stay registered, which will wake the task unnecessarily every time something happens to the socket, again, forever.

TCP is a stream, so polling TCP socket only once does not make much sense. 
Moreover, in multithreaded program, it is hard to set the waker back again.

Dirbaio, 2022-06-06:
Closing due to inactivity. If you have more questions, feel free to open new issues!

closed 2022-06-06 by Dirbaio